leggedrobotics/darknet_ros #247: Segmentation fault running darkner_ros_gdb.launch

nikopoulospet, 2020-06-16:
I am using a Jetson nano, with ros melodic, Opencv 3.3.1, CUDA 10.0.
I am unable to launch darknet, it fails with a segmentation fault at the first spin of the node. here is the error.

Waiting for image.
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7f49a7f430 (LWP 27101)]

Thread 1 "darknet_ros" received signal SIGSEGV, Segmentation fault.
0x0000007fb7c3bf98 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/melodic/lib/libroscpp.so
(gdb) bt
#0  0x0000007fb7c3bf98 in ros::CallbackQueue::callAvailable(ros::WallDuration) () at /opt/ros/melodic/lib/libroscpp.so
#1  0x0000007fb7c8fdec in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () at /opt/ros/melodic/lib/libroscpp.so
#2  0x0000007fb7c79a48 in ros::spin() () at /opt/ros/melodic/lib/libroscpp.so
#3  0x000000555555d330 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/ardug/catkin_ws/src/darknet_ros/darknet_ros/src/yolo_object_detector_node.cpp:17

I tried disabling cuda in the CMAKE file with no luck.

tomlankhorst, 2020-06-16:
Have you tried to attach a debugger to locate the exact source of the error?

nikopoulospet, 2020-06-24:
changing to boost version 1.55 solved my issues

closed 2020-06-24 by nikopoulospet

cheexuu, 2023-01-06:
I meet this issue after I install boost 1.81. Thus, I remove boost 1.81. Then, it works. Hope this can help you.