leggedrobotics/darknet_ros #248: CPU real-time testing problem

antondim, 2020-06-19:
Hello, i'm testing Yolov2-tiny only on CPU (also set GPU = 0),  both on bag files and real time camera streams.

At first I cloned the [leggedrobotics repo](https://github.com/leggedrobotics/darknet_ros) which resulted in asynchronized messages between the bounding boxes and detection images.

 What I did to resolve this (as it was recommended on a different issue post) was to assign "boundingboxResults_.image_header.stamp" to the "boundingboxResults_.header.stamp" , which solved my problem **only for bag files**, so I had to find a solution for the real time asynchronization. 

Issue posts (again) led me [here](https://github.com/MayurT/darknet_ros/tree/fix/action_server_result). This version of YoloObjectDetector.cpp actually bypassed the asynchronization.

By modifying .cfg file, Im testing the detector on **320*320 network input resolution** which results almost in **~1 FPS** on **Intel i7-8565u CPU @ 1,8GHz × 8**.
**Q1:** Any rational explanation for bag files initial synchronization solution , not working for for real time streams?
**Q2:** Are my FPS values normal?
**Q3:** Can FPS increase be achieved without modifying the network's input image resolution (which I know that works)? 
**Q4:** OpenMP = 1 or 0 inside /darknet_ros/darknet/Makefile seems to not affect the FPS. Is this really how Multithreading is enabled?

tomlankhorst, 2021-05-19:
Sorry for late reply

Q1: @mbjelonic might now
Q2: On CPU that might be normal. Did you build in release mode?
Q3: Build in release mode and try to tweak the source code to be more efficient
Q4: I cannot confirm this at this point as most people are using their GPU, but you could validate by checking the CPU core utilization when running. If approx. 100% it's single threaded, if > 100% it's multithreaded. 

> Sorry for long post!
No excuse needed.

foxchys, 2021-06-24:
Q4: I think the "darknet_ros" didn't use the OPENMP default.  You could insert the code into the file "darknet_ros/darknet_ros/CMakeLists.txt", if you want to use the openmp in your program.
`
find_package( OpenMP REQUIRED)
if(OPENMP_FOUND)
    message("OPENMP FOUND")
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
    set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} ${OpenMP_EXE_LINKER_FLAGS}")
endif()
`