leggedrobotics/darknet_ros #250: How to use own detection objects

thejeshk, 2020-07-09:
Hello Guys, 

Does any of you have implemented darknet_ros for own objects? 

I am having trouble to use own objects. Can anyone guide me how to do that?

Thank you :)

thejeshk, 2020-07-09:
Hello Guys, 

I tried adding my weights and config file accordingly. But I am getting this error 

Loading weights from /home/tk/X950R_DL/AirSim/ros/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov3_custom_last.weights...Done!
Waiting for image.
[darknet_ros-1] process has died [pid 7110, exit code -11, cmd /home/tk/X950R_DL/AirSim/ros/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=airsim_node/car_1/front_right_custom/Scen __name:=darknet_ros __log:=/home/tk/.ros/log/70dc4652-c1e4-11ea-9d80-c8d3ff439acd/darknet_ros-1.log].
log file: /home/tk/.ros/log/70dc4652-c1e4-11ea-9d80-c8d3ff439acd/darknet_ros-1*.log

Could anyone have the similar issue? Or If any of you have an indea how to debug this issue please guide me. 

Thank you 
TK

Shreyas0812, 2020-07-11:
Hey, this is not regarding the answer, but can you please guide me, in what command you ran for custom object detection??

thejeshk, 2020-07-14:
Hi @Shreyas0812 

It is working now for my own custom objects

closed 2020-07-14 by thejeshk

renowator, 2020-08-30:
Hello @thejeshk I have the same problem. How did you solve this?

hsaleem1, 2020-10-12:
@thejeshk, can you please guide me with the procedure that how can I train and test custom objects with this package?

Thanks for your help.

renowator, 2020-10-12:
@hsaleem1  I dont think this package provides training  code. You need to train your own YOLO weights somewhere else and then import them into this package to perform detection