leggedrobotics/darknet_ros #254: QObject::~QObject: Timers cannot be stopped from another thread

zuoligang1997, 2020-07-31:
CUDA Error: invalid device symbol
CUDA Error: invalid device symbol: 成功
QObject::~QObject: Timers cannot be stopped from another thread
[darknet_ros-1] process has died [pid 13909, exit code 255, cmd /home/hello/catkin_lll/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/camera/rgb/image_raw __name:=darknet_ros __log:=/home/hello/.ros/log/ed63706a-d30f-11ea-9003-b068e699d999/darknet_ros-1.log].
log file: /home/hello/.ros/log/ed63706a-d30f-11ea-9003-b068e699d999/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
**I run the realsense camera node, and I get an error while running this node, how can I solve it?**

kubs0ne, 2020-08-12:
Check if You have all the necessary files related to YOLO models. So for example: yolo_v4.weights, yolo_v4.cfg and yolo_v4.yaml.

tomlankhorst, 2021-05-19:
I hope you were able to resolve the issue. If not, feel free to reopen.

closed 2021-05-19 by tomlankhorst