leggedrobotics/darknet_ros #255: Cannot open darknet_ros/detection_image in rqt_image_viewer

kubs0ne, 2020-08-12:
Hi,
I'm trying to integrate ROS with YOLO, and I want to be able to play the darknet_ros/detection_image topic from bag and visualise it. The problem is that when I open rqt_image_viewer and select this topic, some artefacts are visible. I tried to change formats in YoloObjectDetector.cpp but it didn't help. I'm using yolo_v4.launch to initialize. What can I do to resolve the issue? Maybe I'm using the wrong visualiser (rqt_image_viewer)?
I would be grateful for any help.

MingheCao, 2020-09-17:
Same issue. The images cannot be visualized by rqt_image_viewer when I set 'enable_opencv' to false. But works fine when opencv window displayed.

RDaneelOlivav, 2021-03-01:
Same issue here. Getting these strange artifacts when launching the yolo_v3.launch

[image]

Tried alsoin RVIZ and same issue. Any ideas? Is there something wrong:

bool YoloObjectDetector::publishDetectionImage(const cv::Mat& detectionImage)
{
  if (detectionImagePublisher_.getNumSubscribers() < 1)
    return false;
  cv_bridge::CvImage cvImage;
  cvImage.header.stamp = ros::Time::now();
  cvImage.header.frame_id = "detection_image";
  cvImage.encoding = sensor_msgs::image_encodings::BGR8;
  cvImage.image = detectionImage;
  detectionImagePublisher_.publish(*cvImage.toImageMsg());
  ROS_DEBUG("Detection image has been published.");
  return true;
}

?

idochi, 2022-04-11:
I use rviz and just have a frozen image from my camera with no bounding boxes.

huangxiaoxiao-5, 2022-08-16:
> 我使用rviz，只是从我的相机中获得了一个没有边界框的冻结图像。

me too,Did you solve the problem？

SergioFerrando, 2023-04-24:
mee to, did you know how to solve the problem?

GavinGaoYan, 2023-04-25:
> mee to, did you know how to solve the problem?

Just disable the darknet's UI. Then it will be OK.

SergioFerrando, 2023-04-25:
how do i do it?

GavinGaoYan, 2023-04-25:
> how do i do it?

in  config/ros.yaml

`image_view:

  enable_opencv: true
  wait_key_delay: 1
  enable_console_output: true`

set enable_opencv=false