leggedrobotics/darknet_ros #256: Oh, my god, who can help me！Using nvidia jetson TX2, but I have only 0.7 FPS

alexis-pepe, 2020-08-15:
Hello, I am also using TX2 to run Darknet_ros, combined with Yolov3.Weights and using the command roslaunch darknet_ros darknet_ros.Launch. FPS is only about 0.7, When I use Yolov2-tiny weights, FPS increases to about 3.7FPS, but the effect is still not ideal.

My camera is Xbox kinectv1, Ubuntu16.04, I use cuda8 and cudnn5.1.1. the command nvcc -V can display the version number normally. I build in Release mode. And I edit the Makefile: GPU=1 OPENCV=1 CUDNN=1 ,when darknet_ros is working,  GPU-Util is 99%.

I looked for relevant questions in "Issues" and found a strange place, I check the output of CMake when building darknet_ros. I noticed that there was nothing behind "CUDA Version:", but "CUDA Libararies:" gave the right path to cuda, Is it normal to display something like this? I've tried so many ways, I'm going crazy!!!

Please help me @tomlankhorst @mbjelonic

alexis-pepe, 2020-08-17:
I noticed that in the makefile, whether GPU and CUDNN are set to 1 and FPS are the same, only 0.3 FPS for Yolov3 and only 3Fps for Yolov3-tiny, GPU-Util is 99%.  I correctly installed cuda8.0.64, and cudnn5.1.10. What's the reason?I've been looking for all sorts of reasons. Oh, my God!

tomlankhorst, 2020-08-18:
Dear @alexis-pepe , sorry for not being able to help you right away now. Hope to able later. Have related issues been able to help you? https://github.com/leggedrobotics/darknet_ros/issues?q=is%3Aissue+fps

closed 2020-09-15 by alexis-pepe