leggedrobotics/darknet_ros #258: How can I estimate distance from object to my camera based on YOLO? (I have both stereo and 2D camera)

thejeshk, 2020-08-26:
Hi, 

I am using darknet_ros for custom object detection and I have one class called "traffic_cone" 

Now, I am able to detect objects but I need to get the distance from the object detected. Is there anyway to do it? 

I have gone through this thread but no success!! https://github.com/leggedrobotics/darknet_ros/issues/103

Could anyone help me how to achieve this?

Thank you :)
TK

renowator, 2020-08-30:
Hey, I guess there are some techniques to get distance with stereo camera, for  using 2D camera you can do this: https://www.youtube.com/watch?v=h8WoErQle2U

Alex-Beh, 2020-08-31:
@thejeshk I use #103 as reference and write a package to retrieve the real distance. You can find the related codes in the following link:
[**link**](https://github.com/Alex-Beh/realsense_yolo/blob/9944c21530c6a515e27ea3abb660246f9c25195d/src/realsense_yolo.cpp#L356)

I retrieve the camera_info once in [**link**](https://github.com/Alex-Beh/realsense_yolo/blob/9944c21530c6a515e27ea3abb660246f9c25195d/src/realsense_yolo.cpp#L67)

Blackcipher101, 2020-10-04:
You can try OpenCV depth function