leggedrobotics/darknet_ros #260: I train my own yolov3-tiny.weights,but no object be detected.

jack162, 2020-09-01:
Hello everyone,

I use ./darknet detector train button-tiny/cfg/obj.data button-tiny/cfg/yolov3-tiny.cfg darknet53.conv.74 to train my own weights.
here are my cfg file

**obj.data**
classes= 8
train  = /home/josh/darknet/button-tiny/cfg/train.txt
valid  = /home/josh/darknet/button-tiny/cfg/test.txt
names = /home/josh/darknet/button-tiny/cfg/obj.names
backup = /home/josh/darknet/button-tiny/cfg/weights/

**obj.names**
up
down
3
2
1
B1
close
open

**yolov3-tiny.cfg**
batch=24
subdivisions=8
classes=8
filter=(classes+3)*5=29

then i get yolov3-tiny_40000.weights

_it is function in darknet that i test it by my computer camear_
./darknet detector demo button-tiny/cfg/obj.data button-tiny/cfg/yolov3-tiny.cfg yolov3-tiny_40000.weights
the outcome looks like this
[image]

_but then i put my weights to darknet_ros,the buttons can not be detect._
i chang network_param_file in  yolo_v3.launch to yolov3-tiny-button.yaml.

**yolov3-tiny-button.yaml.**
yolo_model:

  config_file:
    name: yolov3-tiny.cfg
  weight_file:
    name: yolov3-tiny_40000.weights
  threshold:
    value: 0.3
  detection_classes:
    names:
      - up
      - down
      - 3
      - 2
      - 1
      - B1
      - close
      - open

By the way,the yolov3-tiny.cfg is same as the I use for training.only  batch and
subdivisions i set to 1.

if anyone face to same problem?
thanks a lot!

FamiliennameistChow, 2020-11-10:
i meet the same problem, have you solved it

FamiliennameistChow, 2021-04-07:
maybe you can change you yaml file to

yolov3-tiny-button.yaml.
yolo_model:

config_file:
name: yolov3-tiny.cfg
weight_file:
name: yolov3-tiny_40000.weights
threshold:
value: 0.3
detection_classes:
names:
- up
- down
- F3
- F2
- F1
- B1
- close
- open

it seems that if there are Arabic numerals in the ”names：“, there will be some mistake

ican9595, 2021-05-06:
Hey @jack162, 
is it possible to train your yolov3-tiny model with darknet53.conv.74? Dont you need the yolov3-tiny.conv.15 and how does your yolov3-tiny.cfg file look like?

SanghyunPark01, 2021-11-15:
Hi @jack162 ,
I meet the same problem, have you solved it?