leggedrobotics/darknet_ros #262: How can i get distance and x,y coordinates from object?

godyun7, 2020-09-24:
Hi I'm new in ROS for project. I use darknet_ros yolo. I want to get distance and x,y coordinates. but if i launch yolo_v3.launch file 
[image]. I can't get any coordinate information from object and bounding boxes. how can i get coordinates information. please teach me detail! please please

fgonzalezr1998, 2020-10-22:
Hi @godyun7 I recommend you to use [this package](https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d)

tomlankhorst, 2021-06-08:
Subscribe to the bounding box publisher and determine the values there.

closed 2021-06-08 by tomlankhorst