leggedrobotics/darknet_ros #265: darknet_ros catkin_make issue

mpheng666, 2020-10-01:
Hi I have problem with caktin_make the darknet_ros pkg, any idea how to solve it?

lycoris@lycoris-work-station:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release
Base path: /home/lycoris/catkin_ws
Source space: /home/lycoris/catkin_ws/src
Build space: /home/lycoris/catkin_ws/build
Devel space: /home/lycoris/catkin_ws/devel
Install space: /home/lycoris/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/lycoris/catkin_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/lycoris/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _subt_msgs_generate_messages_check_deps_PoseFromArtifact
[  0%] Built target _subt_msgs_generate_messages_check_deps_SetPose
[  0%] Built target _subt_msgs_generate_messages_check_deps_Bind
[  0%] Built target _subt_msgs_generate_messages_check_deps_Unregister
[  0%] Built target _subt_msgs_generate_messages_check_deps_Register
[  0%] Built target _subt_msgs_generate_messages_check_deps_DatagramRos
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target flipper_control_plugin
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target laser_rotate_plugin
[  1%] Built target subt_rf_range_model
[  1%] Built target rosgraph_msgs_generate_messages_py
[  1%] Built target roscpp_generate_messages_py
[  1%] Built target roscpp_generate_messages_nodejs
[  1%] Built target subt_communication_broker_protobuf_compilation
[  1%] Built target roscpp_generate_messages_eus
[  1%] Built target _x2_msgs_generate_messages_check_deps_Drive
[  1%] Built target roscpp_generate_messages_lisp
[  1%] Built target rosgraph_msgs_generate_messages_cpp
[  1%] Built target roscpp_generate_messages_cpp
[  1%] Built target rosgraph_msgs_generate_messages_nodejs
[  1%] Built target _x2_msgs_generate_messages_check_deps_Feedback
[  1%] Built target rosgraph_msgs_generate_messages_eus
[  1%] Built target _x2_msgs_generate_messages_check_deps_Status
[  1%] Built target _x2_msgs_generate_messages_check_deps_DriveFeedback
[  1%] Built target rosgraph_msgs_generate_messages_lisp
[  2%] Built target protobuf_compilation
[  4%] Built target level_generator
Unpacking cave_circuit_practice_02.dat.zip
[  4%] Built target log_checker
[  6%] Built target dot_generator
x cave_circuit_practice_02.dat
[  6%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionGoal
[  6%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsAction
[  7%] Built target world_generator_cave
[  7%] Built target _darknet_ros_msgs_generate_messages_check_deps_ObjectCount
[  7%] Built target sensor_msgs_generate_messages_py
[  7%] Built target actionlib_msgs_generate_messages_py
[  7%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionFeedback
[  7%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsResult
[  7%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBoxes
[  7%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionResult
[  7%] Built target sensor_msgs_generate_messages_cpp
[  7%] Built target actionlib_msgs_generate_messages_cpp
[  7%] Built target actionlib_msgs_generate_messages_nodejs
[  7%] Built target sensor_msgs_generate_messages_nodejs
[  7%] Built target sensor_msgs_generate_messages_lisp
[  7%] Built target actionlib_msgs_generate_messages_eus
[... 205 lines cut ...]
[ 68%] Building CXX object subt/subt_ros/CMakeFiles/pose_tf_broadcaster.dir/src/pose_tf_broadcaster.cpp.o
[ 68%] Building CXX object subt/subt_ros/CMakeFiles/bridge_logger.dir/src/BridgeLogger.cc.o
[ 69%] Building CXX object subt/subt_ros/CMakeFiles/set_pose_relay.dir/src/SetPoseRelay.cc.o
[ 70%] Building CXX object subt/subt_ros/CMakeFiles/subt_ros_relay.dir/src/SubtRosRelay.cc.o
[ 70%] Building CXX object subt/subt_ign/CMakeFiles/ControllerPlugin.dir/src/ControllerPlugin.cc.o
[ 70%] Building CXX object subt/subt_ign/CMakeFiles/BaseStationPlugin.dir/src/BaseStationPlugin.cc.o
[ 70%] Linking CXX executable /home/lycoris/catkin_ws/devel/lib/subt_ros/pose_tf_broadcaster
[ 70%] Built target pose_tf_broadcaster
[ 71%] Linking CXX executable /home/lycoris/catkin_ws/devel/lib/subt_ros/optical_frame_publisher
[ 71%] Built target optical_frame_publisher
[ 72%] Linking CXX executable /home/lycoris/catkin_ws/devel/lib/subt_ros/bridge_logger
[ 72%] Built target bridge_logger
[ 72%] Linking CXX executable /home/lycoris/catkin_ws/devel/lib/subt_ros/set_pose_relay
[ 73%] Linking CXX shared library /home/lycoris/catkin_ws/devel/lib/libControllerPlugin.so
[ 73%] Built target ControllerPlugin
[ 73%] Built target set_pose_relay
[ 74%] Linking CXX shared library /home/lycoris/catkin_ws/devel/lib/libBaseStationPlugin.so
[ 74%] Built target BaseStationPlugin
[ 74%] Linking CXX executable /home/lycoris/catkin_ws/devel/lib/subt_ros/subt_ros_relay
[ 74%] Built target subt_ros_relay
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

[image]

dharanish-nh, 2020-10-07:
https://github.com/leggedrobotics/darknet_ros/issues/12#issuecomment-310914566

tomlankhorst, 2020-11-27:
Please reopen if the suggested fix did not work.

closed 2020-11-27 by tomlankhorst