leggedrobotics/darknet_ros #266: catkin_make -DCMAKE_BUILD_TYPE=Release

fashlaa, 2020-10-04:
hello everyone I just recursively cloned it. when I want to make I have a problem, even though I have installed CUDA 10.2, CUDNN, and opencv. please help everyone.

Base path: /home/acer/catkin_ws
Source space: /home/acer/catkin_ws/src
Build space: /home/acer/catkin_ws/build
Devel space: /home/acer/catkin_ws/devel
Install space: /home/acer/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/acer/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/acer/catkin_ws/build"
####
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_py
[  2%] Built target talker
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBoxes
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionResult
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsGoal
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_ObjectCount
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsFeedback
[  2%] Built target geometry_msgs_generate_messages_py
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsResult
[  2%] Built target std_msgs_generate_messages_eus
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBox
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsAction
[  2%] Built target sensor_msgs_generate_messages_eus
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionGoal
[  2%] Built target geometry_msgs_generate_messages_eus
[  2%] Built target std_msgs_generate_messages_cpp
[  2%] Built target actionlib_msgs_generate_messages_eus
[  2%] Built target sensor_msgs_generate_messages_cpp
[  2%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionFeedback
[  2%] Built target geometry_msgs_generate_messages_cpp
[  2%] Built target actionlib_msgs_generate_messages_cpp
[  2%] Built target geometry_msgs_generate_messages_nodejs
[  2%] Built target std_msgs_generate_messages_nodejs
[  2%] Built target actionlib_msgs_generate_messages_nodejs
[  2%] Built target sensor_msgs_generate_messages_nodejs
[  2%] Built target sensor_msgs_generate_messages_lisp
[  2%] Built target std_msgs_generate_messages_lisp
[  2%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target actionlib_msgs_generate_messages_lisp
[ 11%] Built target darknet_ros_msgs_generate_messages_eus
[ 18%] Built target darknet_ros_msgs_generate_messages_py
[ 26%] Built target darknet_ros_msgs_generate_messages_cpp
[ 34%] Built target darknet_ros_msgs_generate_messages_nodejs
[ 42%] Built target darknet_ros_msgs_generate_messages_lisp
[ 42%] Built target darknet_ros_msgs_generate_messages
[ 43%] Building CXX object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o
/home/acer/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp: In member function ‘darknet_ros::IplImageWithHeader_ darknet_ros::YoloObjectDetector::getIplImageWithHeader()’:
/home/acer/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:516:49: error: no matching function for call to ‘_IplImage::_IplImage(cv::Mat&)’
   IplImage* ROS_img = new IplImage(camImageCopy_);
                                                 ^
In file included from /usr/local/include/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv2/core/utility.hpp:1234,
                 from /usr/local/include/opencv2/core.hpp:3286,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/melodic/include/cv_bridge/cv_bridge.h:43,
                 from /home/acer/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:30,
                 from /home/acer/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: _IplImage::_IplImage()
 _IplImage
 ^~~~~~~~~
/usr/local/include/opencv2/core/types_c.h:327:1: note:   candidate expects 0 arguments, 1 provided
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(const _IplImage&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘const _IplImage&’
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(_IplImage&&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘_IplImage&&’
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:1956: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o] Error 1
CMakeFiles/Makefile2:2307: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

PLEASE I REALLY NEED HELP, IM STUCK

Antu3heng, 2020-10-10:
Hey, friends. Can you fix it? I have the same problem like you.

fashlaa, 2020-10-15:
> Hey, friends. Can you fix it? I have the same problem like you.

Until now, I still haven't figured out how to fix the problem :(

JitskedeVries, 2020-10-16:
https://github.com/leggedrobotics/darknet_ros/issues/68 this looks like the same problem, I will try it out right now

fashlaa, 2020-10-20:
I have cloned recursively, but every time I want to do make it always gets the same problem 😫

tomlankhorst, 2020-11-27:
Which OS?

tomlankhorst, 2020-11-29:
Closing due to inactivity. Happy to reopen if the problem persists. Please provide more info.

Antu3heng, 2020-11-30:
> Which OS?

ubuntu18.04 OpenCV3.4.11

tomlankhorst, 2020-11-30:
If you are using standard libraries of Ubuntu Bionic, you should be able to build with: 
- recursively cloning the repo
- installing the dependencies (`rosdep install --from-paths src -r -y`)
- `catkin_make -DCMAKE_BUILD_TYPE=Release` or `catkin build -DCMAKE_BUILD_TYPE=Release`

Can you mention anything that's non-standard on your system? 
Could you even reproduce your situation with a Dockerfile for example? 

```
FROM ros:melodic

RUN echo steps to clone and install
```

Antu3heng, 2020-12-01:
I built the code with you steps, but it still had the same problem like this:
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Errors     << darknet_ros:make /home/antusheng/mav_ws/logs/darknet_ros/build.make.001.log 
/home/antusheng/mav_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp: In member function ‘darknet_ros::IplImageWithHeader_ darknet_ros::YoloObjectDetector::getIplImageWithHeader()’:
/home/antusheng/mav_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:516:49: **error: no matching function for call to ‘_IplImage::_IplImage(cv::Mat&)’**
   IplImage* ROS_img = new IplImage(camImageCopy_);
                                                 ^
In file included from /usr/local/include/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv2/core/utility.hpp:1234,
                 from /usr/local/include/opencv2/core.hpp:3284,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/melodic/include/cv_bridge/cv_bridge.h:43,
                 from /home/antusheng/mav_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:30,
                 from /home/antusheng/mav_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: _IplImage::_IplImage()
 _IplImage
 ^~~~~~~~~
/usr/local/include/opencv2/core/types_c.h:327:1: note:   candidate expects 0 arguments, 1 provided
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(const _IplImage&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘const _IplImage&’
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(_IplImage&&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘_IplImage&&’
make[2]: *** [CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o] Error 1
make[1]: *** [CMakeFiles/darknet_ros_lib.dir/all] Error 2
make: *** [all] Error 2
cd /home/antusheng/mav_ws/build/darknet_ros; catkin build --get-env darknet_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
It seems like it's my opencv's error. But I check the opencv file mentioned in the error, I can find the function ‘_IplImage::_IplImage(cv::Mat&)’, but the error said no this function found, which made me confused.

Antu3heng, 2020-12-02:
I think I found the way to fix the problem. The reason why this error showed is the new version opencv doesn't define the macro 'CV__ENABLE_C_API_CTORS'. As a result, the function ‘_IplImage::_IplImage(cv::Mat&)’ would not be defined. 

To fix the problem, just need to add one arg when using the catkin tool to build the darknet_ros `-DCMAKE_CXX_FLAGS=-DCV__ENABLE_C_API_CTORS`. This arg will tell the cmake to define the marco 'CV__ENABLE_C_API_CTORS'.

In short, under the new version opencv, we can build this project with:
`catkin build darknet_ros --cmake-args -DCMAKE_CXX_FLAGS=-DCV__ENABLE_C_API_CTORS `

tomlankhorst, 2020-12-02:
That is of great help. I'll consider adding it to the CMakeLists.txt

tomlankhorst, 2021-05-19:
Issues should be resolved in updated versions.

closed 2021-05-19 by tomlankhorst