leggedrobotics/darknet_ros #272: DARKNET_PATH-NOTFOUND/src/activation_layer.c

kashifzr, 2020-10-18:
<pre>Configuring done
<font color="#EF2929"><b>CMake Error</b></font> at /usr/local/share/cmake-3.14/Modules/FindCUDA.cmake:1813 (add_library):
  Cannot find source file:

    DARKNET_PATH-NOTFOUND/src/activation_layer.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
<font color="#06989A"><u style="text-decoration-style:single">Call Stack (most recent call first):</u></font>
  darknet_ros/darknet_ros/CMakeLists.txt:145 (cuda_add_library)

<font color="#EF2929"><b>CMake Error</b></font> at /usr/local/share/cmake-3.14/Modules/FindCUDA.cmake:1813 (add_library):
  No SOURCES given to target: darknet_ros_lib
<font color="#06989A"><u style="text-decoration-style:single">Call Stack (most recent call first):</u></font>
  darknet_ros/darknet_ros/CMakeLists.txt:145 (cuda_add_library)

-- Build files have been written to: /home/abc/catkin_ws/build
Makefile:320: recipe for target &apos;cmake_check_build_system&apos; failed
make: *** [cmake_check_build_system] Error 1
<font color="#CC0000">Invoking </font><font color="#EF2929"><b>&quot;make cmake_check_build_system&quot;</b></font><font color="#CC0000"> failed</font>
</pre>

tomlankhorst, 2020-11-27:
Have you initialized the darknet git submodule?

tomlankhorst, 2020-12-01:
Please try to recursively checkout the repo as demonstrated in the docs.

closed 2020-12-01 by tomlankhorst