leggedrobotics/darknet_ros #273: catkin_make fails on jetson nano with jetpack 4.4

marcostrullato, 2020-10-18:
Dear all,

I have a fresh installation of JetPack 4.4 on a Nvidia Jetson Nano, which includes opencv 4.1.1 and it's basically an Ubuntu 18.04

After installing ros melodic, ros cv_bridge and few more other packages I checked out the code of the project and built it, but catkin_make fails.

Do you have clue? How do you advise to proceed?

I get the same behavior also specifying the opencv install path. -DOpenCV_INSTALL_PATH=/usr/include/opencv4/

```
marco@roomba3:~/catkin_workspace$ catkin_make -DCMAKE_BUILD_TYPE=Release 
Base path: /home/marco/catkin_workspace
Source space: /home/marco/catkin_workspace/src
Build space: /home/marco/catkin_workspace/build
Devel space: /home/marco/catkin_workspace/devel
Install space: /home/marco/catkin_workspace/install
####
#### Running command: "cmake /home/marco/catkin_workspace/src -DCMAKE_BUILD_TYPE=Release -DOpenCV_INSTALL_PATH=/usr/include/opencv4/opencv2 -DCATKIN_DEVEL_PREFIX=/home/marco/catkin_workspace/devel -DCMAKE_INSTALL_PREFIX=/home/marco/catkin_workspace/install -G Unix Makefiles" in "/home/marco/catkin_workspace/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/marco/catkin_workspace/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/marco/catkin_workspace/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - darknet_ros_msgs
-- ~~  - darknet_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'darknet_ros_msgs'
-- ==> add_subdirectory(darknet_ros/darknet_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action darknet_ros_msgs/CheckForObjects /home/marco/catkin_workspace/src/darknet_ros/darknet_ros_msgs/action/CheckForObjects.action
-- darknet_ros_msgs: 10 messages, 0 services
-- +++ processing catkin package: 'darknet_ros'
-- ==> add_subdirectory(darknet_ros/darknet_ros)
-- Darknet path dir = /home/marco/catkin_workspace/src/darknet_ros/darknet
-- CUDA Version: 
-- CUDA Libararies: /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- Searching for X11...
--  X11_INCLUDE_DIR: /usr/include
--  X11_LIBRARIES: /usr/lib/aarch64-linux-gnu/libSM.so/usr/lib/aarch64-linux-gnu/libICE.so/usr/lib/aarch64-linux-gnu/libX11.so/usr/lib/aarch64-linux-gnu/libXext.so
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   chrono
--   system
--   date_time
--   atomic
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  darknet_ros/darknet_ros/CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/marco/catkin_workspace/build/CMakeFiles/CMakeOutput.log".
See also "/home/marco/catkin_workspace/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

takumi-web, 2020-10-29:
hello!
I have same error on jetson nano and I want to know how to solve this problem.

brsbilgin, 2020-10-29:
Download from github the source code and put it to your workspace. 
https://github.com/ros-perception/vision_opencv/tree/melodic

z760, 2020-11-12:
$cd   at /opt/ros/melodic/share/cv_bridge/cmake/ 
next 
$vim cv_bridgeConfig.cmake
line 96 change opencv  to opencv4

tomlankhorst, 2020-11-27:
Did any of the proposed solutions work? Did you install all dependencies with `rosdep --from-paths src -y -r` ?

marcostrullato, 2020-11-30:
Hi, sorry for the late reply.
I followed the instruction of @brsbilgin and it worked.

Thanks, feel free to close the ticket.

Cheers