leggedrobotics/darknet_ros #274: CUDA Error on Unit Test

dbalaban, 2020-10-23:
I get the error:

`CUDA Error: invalid device symbol
CUDA Error: invalid device symbol: Success
`
when running the unit tests, I have the GPU flag set to 0 in the Makefile in the darknet submodule, and for good measure I commented out the sections of the CMakeLists.txt that use any CUDA packages, so I'm not sure why CUDA is even relevant in this case, or how I can get a CUDA error when it shouldn't be built. The project compiles but the unit tests fails. When I do attempt to compile with CUDA, I get the same results.

here's the output for nvcc:
`nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Nov__3_21:07:56_CDT_2017
Cuda compilation tools, release 9.1, V9.1.85
`

tomlankhorst, 2020-11-27:
Sorry for the late reply. Could you tell me which unit test failed?

dbalaban, 2020-11-27:
I ran the unit tests mentioned in the README verbatum.

`catkin build darknet_ros --no-deps --verbose --catkin-make-args run_tests`

tomlankhorst, 2020-11-28:
Thanks for your reply. 
I see, you ran the unit tests with that command. 

Did the output give an indication of which test failed? 
Did you get more context in another way?

dbalaban, 2020-12-05:
Hi, sorry I couldn't reply sooner, I just ran the test again for the first time since making this post, and the tests succeeded. I'm not sure what changed in that time, but it may have been the inclusion of these lines in the bashrc:

`export LD_LIBRARY_PATH=/usr/local/cuda-11.1/lib64:$LD_LIBRARY_PATH
export PATH=/usr/local/cuda-11.1/bin:$PATH
`

closed 2021-06-08 by tomlankhorst