leggedrobotics/darknet_ros #284: How to disable darknet console output?

zoombinis, 2020-12-20:
In config/ros.yaml I set:

```yaml
image_view:

  enable_opencv: false
  wait_key_delay: 1
  enable_console_output: false
```

But still darknet is producting console output that I do not want. Anyway to stop it?:

>person: 34%
person: 34%
person: 31%
person: 33%
person: 35%
person: 32%
person: 37%
person: 32%
person: 36%
person: 38%
person: 41%
person: 38%

ILG-00, 2021-01-07:
Hi, @zoombinis 
Try commenting this line inside the image.c

`printf("%s: %.0f%%\n", names[j], dets[i].prob[j]*100);`

The file is in darknet/src, and the line is inside draw_detections()

zoombinis, 2021-01-07:
would be great if this were configurable but maintainers are welcome to close this issue if that's not something they wish to support.

For reference, it's `darknet/src/http_stream.cpp` line `609`

ILG-00, 2021-01-07:
I added a `bool console_out` parameter in line `239` of `darknet/src/image.c` and  `735` of `darknet/include/darknet.h` 

And modifid  line `255` in `darknet/src/image.c`
`if (bool console_out) printf("%s: %.0f%%\n", names[j], dets[i].prob[j]*100);`
 
And line `315` of `darknet_ros/darknet_ros/src/YoloObjectDetector.cpp` 
`draw_detections(display, dets, nboxes, demoThresh_, demoNames_, demoAlphabet_, demoClasses_, enableConsoleOutput_);`

Also, some other files (mostly examples) may display error in this function during compilation.

tomlankhorst, 2021-04-12:
Thanks for your feedback. This output is produced by the underlying darknet library and is not configurable there. Consider opening a feature request here: https://github.com/pjreddie/darknet

closed 2021-04-12 by tomlankhorst