leggedrobotics/darknet_ros #285: ModuleNotFoundError: No module named 'darknet_ros_msgs'

Schloern93, 2020-12-21:
Hello

I have built the YOLO-ROS on my Ubuntu 18.04 system with Ros eloquent. 
This works so far. My camera is displayed with the baunding boxes. 
If I now try to display the following Ros Topics with ros2 topic echo cih get the following error:

Topic: 
- /darknet_ros/found_object
- /darknet_ros/bounding_boxes 

Error:
ros2 topic echo /darknet_ros/bounding_boxes 
Traceback (most recent call last):
  File "/opt/ros/eloquent/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.8.8', 'console_scripts', 'ros2')()
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2topic/verb/echo.py", line 83, in main
    return main(args)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2topic/verb/echo.py", line 97, in main
    node.node, args.topic_name, args.message_type, callback, qos_profile)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2topic/verb/echo.py", line 131, in subscriber
    msg_module = import_message_type(topic_name, message_type)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2topic/api/__init__.py", line 74, in import_message_type
    module = importlib.import_module(package_name + '.' + middle_module)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'darknet_ros_msgs'

The following topic works without problems:
- /darknet_ros/detection_image

I have checked the cMake list of the package darknet_ros_msgs. This seems to be ok so far. 

Can someone help me with this? 
Thanks in advance!

closed 2020-12-21 by Schloern93

poodet, 2021-01-08:
HI ! Did you succeed to solve it ?

Schloern93, 2021-01-08:
Yes I have solved it. I had a problem with the source command. Which was a bug in my operating system.
After I solved it I could source the built workspace normally again and display the message with ros2 topic.