leggedrobotics/darknet_ros #286: Displaying the bounding boxes in rviz

Schloern93, 2020-12-22:
Hallo, 

Can the baunding boxes which are specified in pixel be displayed in nrviz?
Or is there the possibility to infer the meters from the pixels?
Sorry for the possibly stupid question, but I am still quite new to the image processing world.

Thanks for the help.

wildpig888, 2021-03-31:
> Hallo,
> 
> Can the baunding boxes which are specified in pixel be displayed in nrviz?
> Or is there the possibility to infer the meters from the pixels?
> Sorry for the possibly stupid question, but I am still quite new to the image processing world.
> 
> Thanks for the help.

I run yolov3 successfully, but I can't run yolo(AB)

tomlankhorst, 2021-05-19:
Object detection works in the image space: the 2D plane of RGB pixels that makes up an image. 
The bounding boxes are defined in pixels too. There is no trivial way to convert these points in image space to coordinates in the 3D world. 
Estimating real-world coordinates from features in images is a lively topic of research though. There might be papers that discuss this topic.

closed 2021-05-19 by tomlankhorst