leggedrobotics/darknet_ros #288: 0 CUDA Error

burak052, 2020-12-27:
`0 CUDA Error: no CUDA-capable device is detected
CUDA Error: no CUDA-capable device is detected: Bad file descriptor
[darknet_ros-1] process has died [pid 15933, exit code -6, cmd /home/zaphkiel/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/zaphkiel/.ros/log/c17b16f2-4839-11eb-aa2e-f0038c4d290f/darknet_ros-1.log].
log file: /home/zaphkiel/.ros/log/c17b16f2-4839-11eb-aa2e-f0038c4d290f/darknet_ros-1*.log

`
how can i fix this error?

[image]

TumVink, 2021-02-05:
Hey. Same issue how to solve it?

tomlankhorst, 2021-02-08:
Does `nvidia-smi` show you any results?

patrick1bauer, 2021-03-24:
nvidia-smi no result for me

RotorBoy2020, 2021-04-14:
I am having this same error as well. I am using VMWare with Ubuntu 18.04. Is there a way to run darknet without needing a CUDA GPU?

leeisack, 2021-04-20:
I'm having the same problem, the nvidia-smi output comes out like this

[image]

tomlankhorst, 2021-04-20:
If you don't have / want to use a NVidia GPU, don't install CUDA on your system. 

Try commenting out line 17 https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/CMakeLists.txt#L17 of the CMakeLists.txt when you get CUDA errors but you don't have a CUDA GPU. 

If you do see output, like @leeisack , then you actually should not get this error. Try running darknet with the following environment variable set:

[`CUDA_VISIBLE_DEVICES=0`](https://developer.nvidia.com/blog/cuda-pro-tip-control-gpu-visibility-cuda_visible_devices/), this will instruct CUDA to only make the first device visible. 

(In a single bash session for example, run `export CUDA_VISIBLE_DEVICES=0`)

enekorodriguezplaza, 2021-05-11:
Hi, I have the same issue. nvidia-smi works, like for @leeisack, but doing hat @tomlankhorst suggested did not work. Any other suggestions?

ThornsOoO, 2021-05-18:
If you don't want to use GPU, you can delete "find_ package(CUDA QUIET)" in Cmakelists

leeisack, 2021-05-18:
I solved this, thank you

enekorodriguezplaza, 2021-05-18:
@leeisack could you please share how you solved it?

leeisack, 2021-05-18:
Well... I don't know we build the same problem first. However, in the case of me, the driver was automatically installed when cuda was installed, but if you do this, the latest driver is installed and the version of cuda is automatically increased. So, I installed the driver and cuda by setting the correct version separately.

tomlankhorst, 2021-05-19:
@leeisack 
Thank you for your reply. Can you maybe share what you mean with "So, I installed the driver and cuda by setting the correct version separately."? Maybe you can provide us with the commands you used.

tayfurg7, 2021-05-24:
i still have the same problem.
[image]
my nvidia-smi;
[image]

leeisack, 2021-05-28:
I didn't use a special method... It was solved by installing the driver according to the gpu version I am using. It seems that cuda and cudnn are installed according to the driver. So, in my case, I wanted to install the old versions of cuda and cudnn, but since the gpu is gtx1080, relatively latest version of the driver was installed and cuda and cudnn were automatically upgraded.
As a result, it was resolved by completely removing cuda, driver, and cudnn, and installing all three by entering the desired version directly. There is nothing I can advise other than this...

tayfurg7, 2021-05-28:
I dont know which version required to be installed according to my gpu.
When i choose GPU as hardware acceleration from notebook settings, gpu is not connected to backend. Can be it caused of this problem?

PankaChakma, 2023-02-20:
> i still have the same problem. [image] my nvidia-smi; [image]

did you solve the issue?