leggedrobotics/darknet_ros #289: catkin_make problem about  libdarknet_ros_lib.so

wbzhang233, 2020-12-29:
Hey, I'm facing a dilemma that

- [ 1] libdarknet_ros_lib.so

`
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsFeedback
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionResult
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsResult
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_ObjectCount
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionGoal
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBoxes
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsGoal
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBox
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsAction
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionFeedback
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[  7%] Built target darknet_ros_msgs_generate_messages_cpp
[ 16%] Built target darknet_ros_msgs_generate_messages_py
[ 24%] Built target darknet_ros_msgs_generate_messages_nodejs
[ 32%] Built target darknet_ros_msgs_generate_messages_lisp
[ 41%] Built target darknet_ros_msgs_generate_messages_eus
[ 41%] Built target darknet_ros_msgs_generate_messages
[ 42%] Linking CXX shared library /home/wbzhang/catkin_yolo/devel/lib/libdarknet_ros_lib.so
/usr/bin/ld: 找不到 -lcuda
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:3733: recipe for target '/home/wbzhang/catkin_yolo/devel/lib/libdarknet_ros_lib.so' failed
make[2]: *** [/home/wbzhang/catkin_yolo/devel/lib/libdarknet_ros_lib.so] Error 1
CMakeFiles/Makefile2:2707: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

`

And my OS is ubuntu18.04 with ROS melodic and OpenCV3.2.0, I don't have a GPU.

I'm really curious about how to cope with this problem, thanks!

xuxuan097503, 2021-03-06:
I've just solved this problem recently，You can try this method：
$ cd /usr/lib
$ sudo ln -s /usr/lib/x86_64-linux-gnu/libcuda.so.455.45.01 libcuda.so,
You need to match your own file name about "libcuda.so.455.45.01" and you can see that in usr/lib.

tomlankhorst, 2021-04-12:
Please try again with the latest release. Thanks.

closed 2021-04-12 by tomlankhorst