leggedrobotics/darknet_ros #290: error "OpenCV 4.x+ requires enabled C++11 support"

Hagaik92, 2021-01-05:
Hello, 
Im trying to run darknet_ros and get this message:

/usr/include/opencv4/opencv2/core/cvdef.h:690:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"

Im using a Jetson Xavier NX with jetpack 4.4.1 
i have tried downgrading my opencv version to 3.4.0 but nothing changed.
any solutions? also, if someone has managed to run darknet_ros over xavier i would like to try his configurations.
Thanks!

bigrobinson, 2021-01-18:
I had the same problem. If you search on this problem in the issues you'll find a lot about it. I solved it by cloning a fork from @kunaltyagi.

```
cd catkin_ws/src
git clone --recursive https://github.com/kunaltyagi/darknet_ros.git
```
Take a look at the available branches, choose build for OpenCV4, and do a recursive update of the submodule to insure you get Darknet.
```
git branch -a
git checkout remotes/origin/opencv4
git submodule update --recursive
```
Finally, build as instructed in the readme.
`catkin_make -DCMAKE_BUILD_TYPE=Release`

kunaltyagi, 2021-01-21:
@bigrobinson I've been having trying to switch to the the AlexyAB fork, which is much more active for the darknet repo, but am having linker issues with that.

I was happy enough with the OpenCV4 port for the original darknet repo, so I just left it there.

WRT performance impacts, **I didn't profile**, but I think there will be no negative impacts due to change from C to C++. With the same code, the performance shouldn't be affected much (or even unaffected)

bigrobinson, 2021-01-29:
Understood @kunaltyagi. I agree performance should not suffer. Thank you so much for your work on this my friend!

Hagaik92, 2021-01-31:
I fixed the problem by rebuilding Cv_bridge and downgrade to OpenCV 3.4.0  . it didnt work with any other versions .

bigrobinson, 2021-02-02:
Update: I have now successfully run the opencv4 branch I mentioned above and confirmed that it works. In my case I used a Basler camera and the pylon ROS drivers.

closed 2021-04-08 by mbjelonic

iamrajee, 2021-10-04:
> I fixed the problem by rebuilding Cv_bridge and downgrade to OpenCV 3.4.0 . it didnt work with any other versions .

Can you eloborate the steps? Thanks!

QuantumY-CHN, 2022-06-04:
@kunaltyagi Thank you for your fork. I tried yours following the steps written by bigrobinson but still got a strange error via catkin_make. 
`/home/quantumy/catkin_wss/catkin_ws_May/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:16:49: error: expected primary-expression before ‘;’ token
   16 | std::string darknetFilePath_ = DARKNET_FILE_PATH;
      |                                                 ^
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:2895: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4108: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

I've checked YoloObjectDetector.cpp several times, and I suppose there's no modification on the original version. Could you give some advice on this error? I appreciate any help you can provide.