leggedrobotics/darknet_ros #292: Versioning

jacotsa, 2021-01-08:
I'm having one heck of a time getting this running, most notably because of an array of opencv install problems.
Can anyone tell me which version of opencv they're running, which works for this package?
I'm not asking for a full list of software compatibility, but I think it would help me and others who would love to use this, to know what everyone's using (successfully)

I can confirm that, although there seems to be some fixes to make opencv 4+ to work, I would advice against using that. 
I'm currently running Ubuntu 20.04. Does anyone have a working version with this?
Is anyone using ros neotic? Or does this exclusively work with melodic

Thanks in advance,

Jaco

jacotsa, 2021-01-10:
OpenCV 3.4.8 seems good

tomlankhorst, 2021-04-09:
It works with Noetic and Melodic now.

closed 2021-04-09 by tomlankhorst