leggedrobotics/darknet_ros #293: Always no output

potBagMeat, 2021-01-14:
Hi developers!

I recently encountered a problem. There were not any outputs for my darknet_ros node. I tried several  networks and weights (all downloaded from cmake process), it was the same situation. When I lowered the threshold , there would be some outputs, but the probability of all the objects were always the same.

hardware environment: Manifold 2-G, Jetson TX2.

software environment: Cuda 9.0+cudnn7.1.4 (arm architecture).

I git clone with recursive option, so it was not the problem.

I also added several lines in my CMakeLists.txt.

in found CUDA area, I added 
`add_definitions(-DCUDNN)`

and also  
`  target_link_libraries(${PROJECT_NAME}_lib
    ...
    cudnn
  )`
there was no any errors in compilation.

anybody know where the problem is?

I will really appreciate it if anyone helps me! : )

3bobo, 2021-01-29:
hi, I met the same question, and I always have no output even when the threshold value is 0 after launch  `darknet_ros.launch`. have you solved it?

potBagMeat, 2021-02-03:
> hi, I met the same question, and I always have no output even when the threshold value is 0 after launch `darknet_ros.launch`. have you solved it?

sorry, still no solution.

tomlankhorst, 2021-04-12:
Have you tried with the latest release?

tomlankhorst, 2021-05-19:
If this is resolved, feel free the share how. If not, feel free to comment on the current state.

closed 2021-05-19 by tomlankhorst