leggedrobotics/darknet_ros #297: no object detected

3bobo, 2021-01-31:
Hi, I met some error when I `roslaunch darknet_ros darknet_ros.launch `, I used `usb_cam` to get the camera image, the topic is `/usb_cam/image_raw`, and I also changed the corresponding topic in `ros.yaml`. The error is, the launch could successfully run, but no object detected, whatever I tried, it only outputs

> FPS:24.5
> Objects:
> 
> 
> 
> 
> FPS:22.9
> Objects:

I check the topic `/usb_cam/image_raw` that it has been corrected sent to `darknet_ros` node, and in rviz it shows the image successfully, therefore I don't know where is the problem.

wildpig888, 2021-03-30:
I met the same problems. Can anyone help?

wildpig888, 2021-03-31:
> Hi, I met some error when I `roslaunch darknet_ros darknet_ros.launch `, I used `usb_cam` to get the camera image, the topic is `/usb_cam/image_raw`, and I also changed the corresponding topic in [...]

Which version did you use, yolov3 or AB?

3bobo, 2021-04-07:
@wildpig888 sorry for reply late, I use yolov3 and it doesn't work. Finally I copy a new version from my friend and it works, thus I still do not know why it does not work at first.

tomlankhorst, 2021-04-09:
You copied what, exactly?

wildpig888, 2021-04-09:
> @wildpig888 sorry for reply late, I use yolov3 and it doesn't work. Finally I copy a new version from my friend and it works, thus I still do not know why it does not work at first.
No detection in darknet_ros, while darknet is detecting with the same .weights #307
You can see if it will work.

3bobo, 2021-04-09:
> You copied what, exactly?

The whole project he downloaded before.

tomlankhorst, 2021-06-10:
Hope your issues are resolved. If you want to know more, please provide detailed info.

closed 2021-06-10 by tomlankhorst