leggedrobotics/darknet_ros #298: fps problem

ctgloglu, 2021-02-02:
when i run it i get 0.1 fps. Could you help me?

mathiasmantelli, 2021-02-16:
I'm also having this issue when using "yolov3.yaml", and 1 fps when using "yolov2-tiny.yaml". My computer has 32Gb, Intel Core i7-8700K 3.7GHzx12, and TITAN Xp/PCie/SSE2, with such computer I wasn't expecting that. If there's anything that I can do to improve that, it would be nice.

ctgloglu, 2021-02-16:
I solved my problem as follows.First, I did the nvidia driver installation. Secondly I installed cuda.After doing these steps I got an error Invoking "make -j8 -l8" failed in my catkin_make.go to this directory src / darknet_ros / darknet_ros / CMakeLists.txt and -gencode arch = compute_30, code = sm_30 remove this line.it started working when i did these steps.I hope it helps your problem.

closed 2021-02-16 by ctgloglu

mathiasmantelli, 2021-02-16:
Hi @ctgloglu 
Thank you for sharing. I'll give it a try and I'll let you know whether it worked. 

Have a nice day

mathiasmantelli, 2021-02-16:
Hi @ctgloglu 

I've followed your recommendations, and it seems CUDA is installed now, but I still have the low fps issue. According to the code below, I have CUDA version 11.2, and CUDA compilation tools V9.1.85:
`mathias@mathiasphi:~$ nvidia-smi
Tue Feb 16 08:52:35 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.32.03    Driver Version: 460.32.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  TITAN Xp            Off  | 00000000:01:00.0  On |                  N/A |
| 23%   39C    P5    21W / 250W |    627MiB / 12192MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1228      G   /usr/lib/xorg/Xorg                 26MiB |
|    0   N/A  N/A      1378      G   /usr/bin/gnome-shell               93MiB |
|    0   N/A  N/A      1618      G   /usr/lib/xorg/Xorg                194MiB |
|    0   N/A  N/A      1765      G   /usr/bin/gnome-shell               48MiB |
|    0   N/A  N/A      2091      G   ...AAAAAAAA== --shared-files       39MiB |
|    0   N/A  N/A      2382      G   ...gAAAAAAAAA --shared-files      214MiB |
|    0   N/A  N/A      5076      G   gnome-control-center                3MiB |
+-----------------------------------------------------------------------------+
mathias@mathiasphi:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Nov__3_21:07:56_CDT_2017
Cuda compilation tools, release 9.1, V9.1.85
`

I also have removed the "gencode arch..." line that you mentioned, and I've compiled the whole project with the following command: 
`catkin_make -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/gcc-6`

There's no compilation error, which makes me believe everything is alright, but the FPS rate is still very low. 

Any help is appreciated.

mathiasmantelli, 2021-02-16:
I found out the problem was the `-DCMAKE_BUILD_TYPE=Release` tag. I compiled the whole project without it, and now the FPS has increased. However, after running the darknet_ros package for a few minutes, my computer just freezes, even the mouse pointer doesn't move. Any ideas why is this happening now?

TapleFlib, 2023-04-24:
Hello , I have fps problem too but I'm using AMD, any idea how can I solve this problem?