leggedrobotics/darknet_ros #303: CMake Error, cannot find: boost_python37

Sacksit, 2021-03-04:
Hi there, I am trying to work with Object detection on a Jetson NX Xavier computer and ran into the issue with boost when doing: catkin_make -DCMAKE_BUILD_TYPE=Release. I'd appreciate any help. The following is the error:

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python37; skipping header check
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

-- Found OpenCV: /usr (found suitable version "4.1.1", minimum required is "4") found components:  opencv_core opencv_imgproc opencv_imgcodecs 
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7") 
-- +++ processing catkin package: 'darknet_ros_msgs'
-- ==> add_subdirectory(darknet_ros/darknet_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action darknet_ros_msgs/CheckForObjects /home/a07/catkin_ws/src/darknet_ros/darknet_ros_msgs/action/CheckForObjects.action
-- darknet_ros_msgs: 10 messages, 0 services
-- +++ processing catkin package: 'image_geometry'
-- ==> add_subdirectory(vision_opencv/image_geometry)
-- Found OpenCV: /usr (found version "4.1.1") 
-- +++ processing catkin package: 'darknet_ros'
-- ==> add_subdirectory(darknet_ros/darknet_ros)
-- Darknet path dir = /home/a07/catkin_ws/src/darknet_ros/darknet
-- CUDA Version: 
-- CUDA Libararies: /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- Searching for X11...
--  X11_INCLUDE_DIR: /usr/include
--  X11_LIBRARIES: /usr/lib/aarch64-linux-gnu/libSM.so/usr/lib/aarch64-linux-gnu/libICE.so/usr/lib/aarch64-linux-gnu/libX11.so/usr/lib/aarch64-linux-gnu/libXext.so
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   chrono
--   system
--   date_time
--   atomic
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Checking and downloading yolov2-tiny.weights if needed ...
-- Checking and downloading yolov3.weights if needed ...
-- Checking and downloading yolov2.weights if needed ...
-- Configuring incomplete, errors occurred!
See also "/home/a07/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/a07/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

tomlankhorst, 2021-04-09:
Can you try with the recently updated version?

tomlankhorst, 2021-05-19:
Feel free to reply.

closed 2021-05-19 by tomlankhorst

sasilva1998, 2021-06-07:
Hello, I have tried recently and have the same problem using the jetson nano.

WAKEYOU, 2021-12-31:
> Hi there, I am trying to work with Object detection on a Jetson NX Xavier computer and ran into the issue with boost when doing: catkin_make -DCMAKE_BUILD_TYPE=Release. I'd appreciate any help. The [...]

Have you solved the problem yet?