leggedrobotics/darknet_ros #306: Don't want the image window to display

Tanguyvans, 2021-03-24:
Hi, 
I am trying to edit the so that I don't have the output image with the bounding boxes. But the rest should work as usual.
How can I do that?

I am using : roslaunch darknet_ros darknet_ros.launch

tomlankhorst, 2021-04-09:
Set the parameter `image_view/enable_opencv` to false.

closed 2021-04-09 by tomlankhorst

zerosnsa, 2024-04-29:
Set the parameter to false.image_view/enable_opencv，What happens when I use the rqt node to view the screen
[image]

zerosnsa, 2024-04-29:
Set this parameter to false. When I use the rqt node to view the screen, the screen display is Mosaic
[image]