leggedrobotics/darknet_ros #307: No detection in darknet_ros, while darknet is detecting with the same .weights

akmaralAW, 2021-03-30:
Hello !

I trained yolo v3 on custom dataset (within darknet). I can see some detections in darknet, however it fails to detect when I use trained  .weights in darknet_ros. 
.cfg, .yaml, .launch are created. 

If anyoane have seen such case, please share your solution.

wildpig888, 2021-03-31:
I downloaded again by the following order.
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git
This time it also downloaded the darknet(yolov3). I make it again and the detections shows successfully.
However, when I tried  run yolo(AB), it still shows no bounding boxes.

wildpig888, 2021-03-31:
I successfully ran YOLO(AB) just now. Did you exchange the darknet folder with your own darknet? Maybe the problem is that. YOU can try tou use git clone to download darknet it provides.

akmaralAW, 2021-04-01:
Oh, great ! So glad to hear it is working for you :)

I am dealing with ssh problem right now, after I will try your solution.  Before I cloned this repository with http, maybe that is why my model is behaving so.

wildpig888, 2021-04-01:
I have tried ssh, but the result was not so good. Now i am learning Socket.

wildpig888, 2021-04-01:
I'm sorry my pool English made you misunderstand what I mean.
I use   'git clone --recursive git@github.com:leggedrobotics/darknet_ros.git' to download the whole code it provides(include Darknet).  Compile directly and it run successfully.  Don't  replace darknet inside draknet_ros with yours. You don't even need to compile Darknet.

akmaralAW, 2021-04-01:
Ura ! It is working :) 
Thanks for your help @wildpig888

closed 2021-04-01 by akmaralAW

tabowsy, 2021-05-30:
@akmaralAW  I did the solution above and successfully detected the object. However, the performance are worse than the original darknet (darknet detect demo). Im wondering if you facing or feeling the same problem? tks

akmaralAW, 2021-05-31:
Hello @tabowsy,

I am afraid my answer won't be helpful, since I didn't test my model within darknet_ros with real images. In a simulated environement it didn't work so well.  I think this is because my original model was not trained well enough, I used just 700 images. 

If you could provide more info on your training part (dataset size, average loss, etc.), I would try to help.

SanghyunPark01, 2021-11-15:
@akmaralAW  I meet same problem. but, i can't solve this problem.  
yolov3 on darknet_ros is doing well. but my model does not work.  
I train my data and get .weights, .cfg. Then, i created .yaml and .launch.  
But mymodel on darknet_ros does not work(no detection, nothing). help....

tabowsy, 2021-11-15:
@SanghyunPark01 I think there's some problem in this repository which need to be fixed. I've tried to use other ros darknet source code and successfully applied my trained weights in it. But it was 5 months ago and I didn't use it for so long, you can search for alternative developments, good luck!

SanghyunPark01, 2021-11-15:
@tabowsy thanks! I will try it