leggedrobotics/darknet_ros #322: Unable to perform catkin_make

devrostech, 2021-05-20:
`[ 79%] Building CXX object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o
In file included from /usr/local/cuda-10.1/include/channel_descriptor.h:61:0,
                 from /usr/local/cuda-10.1/include/cuda_runtime.h:95,
                 from /home/maisumhayati/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:44,
                 from /home/maisumhayati/catkin_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2248:25: error: expected ‘)’ before ‘*’ token
 typedef void (CUDART_CB *cudaStreamCallback_t)(cudaStream_t stream, cudaError_t status, void *userData);
                         ^
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2316:9: error: ‘cudaStreamCallback_t’ has not been declared
         cudaStreamCallback_t callback, void *userData, unsigned int flags);
         ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2483:88: error: use of enum ‘cudaStreamCaptureMode’ without previous declaration
 aError_t CUDARTAPI cudaStreamBeginCapture(cudaStream_t stream, enum cudaStreamCaptureMode mode);
                                                                     ^~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2534:80: error: use of enum ‘cudaStreamCaptureMode’ without previous declaration
 st__ cudaError_t CUDARTAPI cudaThreadExchangeStreamCaptureMode(enum cudaStreamCaptureMode *mode);
                                                                     ^~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2562:81: error: ‘cudaGraph_t’ has not been declared
 t__ cudaError_t CUDARTAPI cudaStreamEndCapture(cudaStream_t stream, cudaGraph_t *pGraph);
                                                                     ^~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2600:87: error: use of enum ‘cudaStreamCaptureStatus’ without previous declaration
 daError_t CUDARTAPI cudaStreamIsCapturing(cudaStream_t stream, enum cudaStreamCaptureStatus *pCaptureStatus);
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:2628:90: error: use of enum ‘cudaStreamCaptureStatus’ without previous declaration
 rror_t CUDARTAPI cudaStreamGetCaptureInfo(cudaStream_t stream, enum cudaStreamCaptureStatus *pCaptureStatus, unsigned long long *pId);
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3012:64: error: ‘cudaExternalMemory_t’ was not declared in this scope
 extern __host__ cudaError_t CUDARTAPI cudaImportExternalMemory(cudaExternalMemory_t *extMem_out, const struct cudaExternalMemoryHandleDesc *memHandleDesc);
                                                                ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3012:64: note: suggested alternative: ‘cudaImportExternalMemory’
 extern __host__ cudaError_t CUDARTAPI cudaImportExternalMemory(cudaExternalMemory_t *extMem_out, const struct cudaExternalMemoryHandleDesc *memHandleDesc);
                                                                ^~~~~~~~~~~~~~~~~~~~
                                                                cudaImportExternalMemory
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3012:86: error: ‘extMem_out’ was not declared in this scope
 udaError_t CUDARTAPI cudaImportExternalMemory(cudaExternalMemory_t *extMem_out, const struct cudaExternalMemoryHandleDesc *memHandleDesc);
                                                                     ^~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3012:98: error: expected primary-expression before ‘const’
 UDARTAPI cudaImportExternalMemory(cudaExternalMemory_t *extMem_out, const struct cudaExternalMemoryHandleDesc *memHandleDesc);
                                                                     ^~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3012:154: error: expression list treated as compound expression in initializer [-fpermissive]
 emory_t *extMem_out, const struct cudaExternalMemoryHandleDesc *memHandleDesc);
                                                                              ^
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3066:88: error: ‘cudaExternalMemory_t’ has not been declared
 aError_t CUDARTAPI cudaExternalMemoryGetMappedBuffer(void **devPtr, cudaExternalMemory_t extMem, const struct cudaExternalMemoryBufferDesc *bufferDesc);
                                                                     ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3121:111: error: ‘cudaExternalMemory_t’ has not been declared
 ExternalMemoryGetMappedMipmappedArray(cudaMipmappedArray_t *mipmap, cudaExternalMemory_t extMem, const struct cudaExternalMemoryMipmappedArrayDesc *mipmapDesc);
                                                                     ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3144:65: error: ‘cudaExternalMemory_t’ was not declared in this scope
 extern __host__ cudaError_t CUDARTAPI cudaDestroyExternalMemory(cudaExternalMemory_t extMem);
                                                                 ^~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3144:65: note: suggested alternative: ‘cudaImportExternalMemory’
 extern __host__ cudaError_t CUDARTAPI cudaDestroyExternalMemory(cudaExternalMemory_t extMem);
                                                                 ^~~~~~~~~~~~~~~~~~~~
                                                                 cudaImportExternalMemory
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3238:67: error: ‘cudaExternalSemaphore_t’ was not declared in this scope
 extern __host__ cudaError_t CUDARTAPI cudaImportExternalSemaphore(cudaExternalSemaphore_t *extSem_out, const struct cudaExternalSemaphoreHandleDesc *semHandleDesc);
                                                                   ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3238:67: note: suggested alternative: ‘cudaImportExternalSemaphore’
 extern __host__ cudaError_t CUDARTAPI cudaImportExternalSemaphore(cudaExternalSemaphore_t *extSem_out, const struct cudaExternalSemaphoreHandleDesc *semHandleDesc);
                                                                   ^~~~~~~~~~~~~~~~~~~~~~~
                                                                   cudaImportExternalSemaphore
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3238:92: error: ‘extSem_out’ was not declared in this scope
 or_t CUDARTAPI cudaImportExternalSemaphore(cudaExternalSemaphore_t *extSem_out, const struct cudaExternalSemaphoreHandleDesc *semHandleDesc);
                                                                     ^~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3238:104: error: expected primary-expression before ‘const’
 PI cudaImportExternalSemaphore(cudaExternalSemaphore_t *extSem_out, const struct cudaExternalSemaphoreHandleDesc *semHandleDesc);
                                                                     ^~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:3238:163: error: expression list treated as compound expression in initializer [-fpermissive]
 re_t *extSem_out, const struct cudaExternalSemaphoreHandleDesc *semHandleDesc);
[... 707 lines cut ...]
                                                                              ^
/usr/local/cuda-10.1/include/cuda_runtime_api.h:9206:60: error: ‘cudaGraphExec_t’ was not declared in this scope
 extern __host__ cudaError_t CUDARTAPI cudaGraphExecDestroy(cudaGraphExec_t graphExec);
                                                            ^~~~~~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:9206:60: note: suggested alternative: ‘cudaGraphCreate’
 extern __host__ cudaError_t CUDARTAPI cudaGraphExecDestroy(cudaGraphExec_t graphExec);
                                                            ^~~~~~~~~~~~~~~
                                                            cudaGraphCreate
/usr/local/cuda-10.1/include/cuda_runtime_api.h:9226:56: error: ‘cudaGraph_t’ was not declared in this scope
 extern __host__ cudaError_t CUDARTAPI cudaGraphDestroy(cudaGraph_t graph);
                                                        ^~~~~~~~~~~
/usr/local/cuda-10.1/include/cuda_runtime_api.h:9226:56: note: suggested alternative: ‘cudaError_t’
[... cut ...]
CMakeFiles/Makefile2:8541: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
`

Please help me building this repository, I want to run the algorithm through GPU acceleration. I am able to perform the **make** command in the darknet folder (catkin_ws/src/darknet_ros/darknet) but I cannot build the entire workspace

tomlankhorst, 2021-05-27:
Make sure to have the CUDA SDK installed. What's your OS?

devrostech, 2021-06-01:
Ubuntu 18.04 with ROS melodic

tomlankhorst, 2021-06-08:
Did you install multiple CUDA versions? Can you update to 10.2?

devrostech, 2021-06-08:
I have resolved the error with cuda 11.3 and cudnn 8.x

closed 2021-06-08 by devrostech

tomlankhorst, 2021-06-08:
Thanks for letting us know.