leggedrobotics/darknet_ros #324: Invoking "make -j4 -l4" failed

MART98, 2021-05-27:
Hello all ... I need help...

Ubuntu 18.04
ROS Melodic

when catkin_make :

.
.
.
[ 67%] Built target turtlebot3_example_generate_messages_eus
[ 73%] Built target turtlebot3_example_generate_messages_lisp
[ 75%] Built target realsense_gazebo_plugin
[ 82%] Built target turtlebot3_example_generate_messages_py
[ 88%] Built target turtlebot3_example_generate_messages_cpp
make[2]: *** No rule to make target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build'.  Stop.
CMakeFiles/Makefile2:3146: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 88%] Built target darknet_ros_msgs_generate_messages
[ 94%] Built target turtlebot3_example_generate_messages_nodejs
[ 94%] Built target realsense2_camera_generate_messages
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

tomlankhorst, 2021-05-27:
Please follow the instructions in the readme. Make sure to clone recursively.

tomlankhorst, 2021-05-28:
If you want help you need to formulate a clear question and provide context: what did you do; how can we reproduce your problem. Thanks for your understanding.

closed 2021-05-28 by tomlankhorst