leggedrobotics/darknet_ros #326: Invoking "make -j12 -l12" failed

konnoeric, 2021-05-30:
**Describe the bug**
A clear and concise description of what the bug is.
I cannot catkin_make. There is an error. How can I fix this?

**To Reproduce**
Steps to reproduce the behavior:
I git cloned using SSH and added --recursive  option ofcourse.
But when I do "catkin_make -DCMAKE_BUILD_TYPE=Release", Invoking "make -j12 -l12" failed
will be showed.

**Expected behavior**
Succecfully catkin_make.

**Screenshots**
[image]
[image]
[image]
[image]

**System specs (please complete the following information):**
Please look at screeshots

**Additional context**
Add any other context about the problem here.
[ 46%] Built target darknet_ros_msgs_generate_messages
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
In file included from /usr/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
CMake Error at darknet_ros_lib_generated_blas_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_blas_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:105: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_blas_kernels.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMake Error at darknet_ros_lib_generated_crop_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_crop_layer_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:84: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_crop_layer_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_im2col_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_im2col_kernels.cu.o

CMake Error at darknet_ros_lib_generated_maxpool_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_maxpool_layer_kernels.cu.o

CMake Error at darknet_ros_lib_generated_activation_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_activation_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:126: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_im2col_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:140: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_maxpool_layer_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:77: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_activation_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_dropout_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_dropout_layer_kernels.cu.o

CMake Error at darknet_ros_lib_generated_deconvolutional_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_deconvolutional_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:98: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_deconvolutional_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_col2im_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_col2im_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:112: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_dropout_layer_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_convolutional_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_convolutional_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:119: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_col2im_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:133: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_convolutional_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_avgpool_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_avgpool_layer_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:91: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_avgpool_layer_kernels.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2505: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j12 -l12" failed

konnoeric, 2021-06-02:
Does anyone know how to fix this?

tomlankhorst, 2021-06-08:
Did you clone recursively the latest release? 
Did CMake output any errors? 
What's your CMake version `cmake --version`.

konnoeric, 2021-06-09:
Yes of course I did git clone recursively with the latest one. 

[image]

The CMake errors are mainly 2 kinds. 
① 
#error -- unsupported GNU version! gcc versions later than 8 are not supported!

②
「CMake Error at darknet_ros_lib_generated_avgpool_layer_kernels.cu.o.Release.cmake:220 (message):
Error generating
/home/konno/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_avgpool_layer_kernels.cu.o」

konnoeric, 2021-06-09:
Also I tried this command and tried to install all of the dependencies but I could not install it.
> sudo rosdep install -yr --from-paths 

[image]

tomlankhorst, 2021-06-10:
Looks like you have an incompatible (outdated) version of CUDA on your system.
Can you try with v. 11 (or 10.2 otherwise).

konnoeric, 2021-06-12:
I installed cuda 11.3 but still cannot catkin_make.
I could not 10.2 on Ubuntu20.04 because it was not available. There were no options for 20.04. Only for 18.04 and 16.04.
[image]

konnoeric, 2021-06-12:
Actually I have already downloaded darknet from [the original site](https://pjreddie.com/darknet/install/) and it is located right next to my catkin_ws in home directory as you can see the screenshot. Is this causing conflicts with your darknet on ROS?

[image]

tomlankhorst, 2021-06-14:
If these directories are not actually in your workspace you should be fine. 
If your workspace is `catkin_ws`, then the workspace source is in `catkin_ws/src`.
This is strange. Looks like it has to do with your CUDA setup. 

You might want to retry in a Docker container so that you have a clean and clear starting point.

yangqiao95, 2021-11-22:
> 您是否递归克隆了最新版本？ CMake 是否输出任何错误？ 你的 CMake 版本是什么`cmake --version`。

Base path: /home/yangq871/catkin_ws
Source space: /home/yangq871/catkin_ws/src
Build space: /home/yangq871/catkin_ws/build
Devel space: /home/yangq871/catkin_ws/devel
Install space: /home/yangq871/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/yangq871/catkin_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/yangq871/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsFeedback
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_ObjectCount
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionResult
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBox
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsAction
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_BoundingBoxes
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionFeedback
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsGoal
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsActionGoal
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _darknet_ros_msgs_generate_messages_check_deps_CheckForObjectsResult
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[ 24%] Built target darknet_ros_msgs_generate_messages_lisp
[ 24%] Built target darknet_ros_msgs_generate_messages_cpp
[ 24%] Built target darknet_ros_msgs_generate_messages_py
[ 32%] Built target darknet_ros_msgs_generate_messages_nodejs
[ 40%] Built target darknet_ros_msgs_generate_messages_eus
[ 40%] Built target darknet_ros_msgs_generate_messages
[ 97%] Built target darknet_ros_lib
[100%] Built target darknet_ros_nodelet
[100%] Linking CXX executable /home/yangq871/catkin_ws/devel/lib/darknet_ros/darknet_ros
/usr/bin/x86_64-linux-gnu-ld: CMakeFiles/darknet_ros.dir/src/yolo_object_detector_node.cpp.o: relocation R_X86_64_32 against symbol `__pthread_key_create@@GLIBC_2.2.5' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/x86_64-linux-gnu-ld: 最后的链结失败: 输出不可表示的节
collect2: error: ld returned 1 exit status
darknet/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:192: recipe for target '/home/yangq871/catkin_ws/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/home/yangq871/catkin_ws/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:2975: recipe for target 'darknet/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

please help me，Thank you
why  catkin_make 100% does  have error? 
g++-6.5，gcc-6.5  ，opencv-4.0，

hamdykhfagy, 2023-10-29:
> Base path: /home/yangq871/catkin_ws Source space: /home/yangq871/catkin_ws/src Build space: /home/yangq871/catkin_ws/build Devel space: /home/yangq871/catkin_ws/devel Install space: [...]

were you able to solve it?

yangqiao95, 2023-11-01:
I  have dealed with  this problem

yangqiao95, 2023-11-01:
thank you for your response