leggedrobotics/darknet_ros #329: libdarknet_ros_lib.so:undefined reference

SuperrHaze, 2021-06-16:
/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `load_alphabet_with_file'
/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `pull_implicit_layer'
/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `push_implicit_layer'
/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `make_implicit_layer'
/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `generate_image'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/haze/catkin_workspace/devel/.private/darknet_ros/lib/darknet_ros/darknet_ros] Error 1
make[1]: *** [CMakeFiles/darknet_ros.dir/all] Error 2
make: *** [all] Error 2
Is there any solution to solve this problem?

SuperrHaze, 2021-06-16:
It seems I have to add sentences like 
"${DARKNET_PATH}/src/image_opencv.cpp"
in "CMakeLists.txt"
"image_opencv.cpp" should be the cpp.file contain the function like "pull_implicit_layer"

SuperrHaze, 2021-06-17:
I solve this problem. I make yolov4 with darknet_ros.
It is because AlexAB added [empty]/[silence] and [implicit] layers on 12 May 
and from the history i know new files called "representation_layer.h" and "representation_layer.c" were created
so I have to add ${DARKNET_PATH}/src/representation_layer.c in CMakefile.list

tomlankhorst, 2021-06-18:
Does this issue occur with the most recent version of this repo?

SuperrHaze, 2021-06-18:
> Does this issue occur with the most recent version of this repo?

I believe so.

xxEoD2242, 2021-07-30:
> I solve this problem. I make yolov4 with darknet_ros.
> It is because AlexAB added [empty]/[silence] and [implicit] layers on 12 May
> and from the history i know new files called [...]

This was my problem as well after I updated some of the code for v4 as well.

johnny12150, 2021-10-19:
> I solve this problem. I make yolov4 with darknet_ros. It is because AlexAB added [empty]/[silence] and [implicit] layers on 12 May and from the history i know new files called [...]

This still works for the latest darkent. Thanks!