leggedrobotics/darknet_ros #330: BGRA8 Support Was Dropped

Halfwake, 2021-06-16:
**Describe the bug**

This wrapper used to support BGRA8 images, but now it doesn't. Before, this library would silently coerce BGRA8 to BGRA.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone`
2. `git checkout 913383f76c81dacc43307dc5cea1f12f5d67835a`
3. `rosrun ros_darknet ros darknet_ros.launch`
4. Send a RGBA8 image.
`#!/usr/bin/env python

import os.path

import rospy
import rospkg
import cv2 as cv
from cv_bridge import CvBridge
from sensor_msgs.msg import Image
from std_msgs.msg import Header

if __name__ == '__main__':
    rospy.init_node('test_bgra8')

    # Prepare Camera Message
    ROS_PACK = rospkg.RosPack()
    BRIDGE = CvBridge()
    TEST_IMAGE = cv.imread(os.path.join(
        ROS_PACK.get_path('virtual_camera_test'),
        'images',
        'some-bgra8-image.whatever'
    ))

    if TEST_IMAGE is None:
        print('Could not load image file.')
        quit()

    TEST_MESSAGE = BRIDGE.cv2_to_imgmsg(
        TEST_IMAGE,
        encoding='bgra8'
    )

    # Prepare Subscriber
    CAMERA_PUBLISHER = rospy.Publisher(
        '/camera/rgb/image_raw',
        Image,
        queue_size=1
    )
    
    print('Starting Virtual Camera Test')
    CAMERA_PUBLISHER.publish(TEST_MESSAGE)

`

See error:
`darknet/src/image_opencv.cpp:14: cv::Mat image_to_mat(image): Assertion `im.c == 3 || im.c == 1' failed.`

**Expected behavior**
I expected a window to open up highlighting YOLO

**System specs (please complete the following information):**
 - OS: Ubuntu 18.0
 - ROS version: melodic
 - Library versions when non-standard
 - GPU / CUDA version when relevant

**Additional context**
This is almost certainly caused by this commit: https://github.com/leggedrobotics/darknet_ros/commit/ae23d0f2fa2fd08b56d61e7aa65c851aa340ae3d

Before, BGRA8 images were silently coerced to BGR8. Now they aren't.

kdhansen, 2021-06-18:
I have been trying to replicate, but I did not get the error. Turns out, this computer does not have CUDA...

However, I had to modify your test code a bit to get a BGRA8 encoding. Seems like cv.imread only reads BGR, no A. @Halfwake, Can you verify that this snippet works on your computer? Then I can test it on my CUDA-enabled computer.

```python
#!/usr/bin/env python

import os.path

import rospy
import rospkg
import cv2 as cv
from cv_bridge import CvBridge
from sensor_msgs.msg import Image
from std_msgs.msg import Header

if __name__ == '__main__':
    rospy.init_node('test_bgra8')

    # Prepare Camera Message
    ROS_PACK = rospkg.RosPack()
    BRIDGE = CvBridge()
    TEST_IMAGE = cv.imread(
        'cam_test.bmp'
    )
    TEST_IMAGE = cv.cvtColor(TEST_IMAGE, cv.COLOR_BGR2BGRA)

    if TEST_IMAGE is None:
        print('Could not load image file.')
        quit()

    TEST_MESSAGE = BRIDGE.cv2_to_imgmsg(
        TEST_IMAGE,
        encoding='bgra8'
    )

    # Prepare Subscriber
    CAMERA_PUBLISHER = rospy.Publisher(
        '/camera/rgb/image_raw',
        Image,
        queue_size=1
    )
    
    print('Starting Virtual Camera Test')
    rate = rospy.Rate(1)
    while not rospy.is_shutdown():
        CAMERA_PUBLISHER.publish(TEST_MESSAGE)
        rate.sleep()
```

Halfwake, 2021-06-18:
I wasn't able to get the script running with the BMP (I guess it couldn't convert the BMP to BGRA?), but I was able to reproduce with another image.
```python
#!/usr/bin/env python

import os.path

import rospy
import rospkg
import cv2 as cv
from cv_bridge import CvBridge
from sensor_msgs.msg import Image
from std_msgs.msg import Header

if __name__ == '__main__':
    rospy.init_node('virtual_camera_test')

    print(__import__('os').getcwd())
    # Prepare Camera Message
    ROS_PACK = rospkg.RosPack()
    BRIDGE = CvBridge()
    TEST_IMAGE = cv.imread(os.path.join(
        ROS_PACK.get_path('virtual_camera_test'),
        'images',
        'dog.jpg'
    ))
    TEST_IMAGE = cv.cvtColor(TEST_IMAGE, cv.COLOR_BGR2BGRA)
    if TEST_IMAGE is None:
        print('Could not load image file.')
        quit()

    TEST_MESSAGE = BRIDGE.cv2_to_imgmsg(
        TEST_IMAGE,
        encoding='bgra8'
    )

    # Prepare Subscriber
    CAMERA_PUBLISHER = rospy.Publisher(
        '/camera/rgb/image_raw',
        Image,
        queue_size=1
    )
    

    # Main Loop
    RATE = rospy.Rate(0.25)
    print('Starting Virtual Camera Test')
    while not rospy.is_shutdown():
        print('Sending Picture')
        CAMERA_PUBLISHER.publish(TEST_MESSAGE)
        RATE.sleep()
```

Image:
[image]

kdhansen, 2021-06-29:
I figured out that it did not have anything to do with CUDA, rather I compiled with -DCMAKE_BUILD_TYPE=Release, which silences the assert statements. Actually, it worked in release mode, with #336 it works in Debug too.

closed 2021-06-30 by mbjelonic