leggedrobotics/darknet_ros #331: YOLOros turns red staffs into blue staffs

SuperrHaze, 2021-06-18:
It seems that red exchanges with blue
[image]
i dont know how to solve it, so i decide to reinstall yoloros and i will comment if this problem is solved

SuperrHaze, 2021-06-18:
I solve this problem。
AlexAB Daknet based on RGB and opencv read pictures on BGR
Alex has already converted BGR into RGB, so darknet works well.
But in YoloObjectDetector.cpp line:377 rgbgr_image(buff_[buffIndex_]);
RGB flips back to BGR,so just notes this sentence and the result turns great.

closed 2021-06-18 by SuperrHaze

johnny12150, 2021-10-21:
> I solve this problem。 AlexAB Daknet based on RGB and opencv read pictures on BGR Alex has already converted BGR into RGB, so darknet works well. But in YoloObjectDetector.cpp line:377 rgbgr_image(buff_[buffIndex_]); RGB flips back to BGR,so just notes this sentence and the result turns great.

I am facing this problem too. 
`rgbgr_image()` turns rgb into bgr and I try to use `cvtColor` with `cv::COLOR_BGR2RGB` to turn the color back in `publishInThread()`.
However, I still get the image of the wrong color that red exchanges into the blue.