leggedrobotics/darknet_ros #335: Ros2 dashing catkim_make fails

ludovicoderic, 2021-06-23:
I tried compiling this package on ROS2 dashing and Ubuntu 18.04, but i couldn't build it.

The steps I did are the following one:
`source /opt/ros/dashing/setup.bash
cd ~/darknet2_ws/src
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git -b ros2
cd ..
colcon build`

And I get the next error:
`Starting >>> Project
--- stderr: Project
CMake Error at CMakeLists.txt:65 (message):
  find_package(catkin) failed.  catkin was neither found in the workspace nor
  in the CMAKE_PREFIX_PATH.  One reason may be that no ROS setup.sh was
  sourced before.

---
Failed   <<< Project [1.80s, exited with code 1]

Summary: 0 packages finished [2.07s]
  1 package failed: Project
  1 package had stderr output: Project`

I tried to clone the repository with the foxy branch and without specifying anything, but i get the same errors.
Also I tried making the project with the -DCMAKE_BUILD_TYPE=Release arg, but with the colcon one I get: `usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,test,test-result,version-check}
              ...
colcon: error: unrecognized arguments: -DCMAKE_BUILD_TYPE=Release`

If I tried to make it with the catkin_make as it saids on the ros2 branch, I get the next error:
`Command 'catkin_make' not found, but can be installed with:

sudo apt install catkin`

How can I get this package running on ROS2 dashing?

closed 2021-06-29 by ludovicoderic

AnukritiSinghh, 2021-07-23:
Hey, how did you solve this? I am facing the same issue

ludovicoderic, 2021-07-23:
In my case it was a problem with the publisher that I had on ROS2 on windows, and the subscriber in WSL Dashing, since I was able to make a ros2 topic echo on WSL and see the data being published, but I was unable for some reason to subscribe to it. That being said, I ended making another publisher that send the data of the image that I needed properly, and to make sure that from a standar topic on ROS2 dashing I was able to subcribe to it and see the image. Once I properly send the image topic, it did work as espected.