leggedrobotics/darknet_ros #341: colcon build failed under foxy : cannot find -ldarknet_ros_core_lib

an99990, 2021-07-16:
**Describe the bug**
Hi,
So I am new to this and trying to build darknet_ros. I have build it under the foxy branch.
I have encoutered many syntax error such as `show_image `not getting enough parameters in attention.c, and `show_image_cv `having declaration conflicts.
I fixed them by either commenting show_image in attention.c and change

```
extern "C" void show_image_cv(image p, const char *name, IplImage *disp) to 
extern "C" void show_image_cv_cp(image p, const char *name, IplImage *disp) in YoloObjectDetector.hpp;
```

I dont know if the changes are related to my error but wanted to let you guys know in case.
here is the error message

```
/usr/bin/ld: cannot find -ldarknet_ros_core_lib
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/darknet_ros.dir/build.make:303: darknet_ros] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/darknet_ros.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Summary: 0 package finished
1 package failed: darknet_ros
1 package had stderr output: darknet_ros

```
**To Reproduce**
Steps to reproduce the behavior:
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

**System (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS version: Foxy
 - GPU ; NVidia GTX 1080

Can anyone help me please ? I have looked and cant find the library ldarknet_ros_core_lib

Ar-Ray-code, 2021-07-18:
What version of Opencv and CUDA are you using... :eyes:  ?

If you are using OpenCV4, Please use OpenCV3 or use my [darknet_ros](https://github.com/Ar-Ray-code/darknet_ros_fp16).

As [my commentary page](https://qiita.com/Ar-Ray/items/93baac6cc716dc1862fe) shows, replacing it with darknet (YOLO v3), which supports OpenCV4, will also solve the problem. (I know it's hard to read in Japanese. Sorry)

an99990, 2021-07-20:
@Ar-Ray-code will your code work with the ZED camera ?

Ar-Ray-code, 2021-07-20:
I'm curious about the message in `LookupError: Could not find the resource 'v4l2_camera' of type 'packages'`. 

Did you run `$ sudo apt install ros-foxy-v4l2-camera` ?

ZED2 Stereo camera is not supported. I want to support it in the future...

Ar-Ray-code, 2021-07-23:
If it is resolved, please close this issue.

kyuhyong, 2022-07-18:
I am having the same error as below
`/usr/bin/ld: cannot find -ldarknet_ros_core_lib`

Have opencv3.4.11 installed with CUDA-11.6
Running Ubuntu 20.04 and apt installed ROS 2 foxy 
Also set -O3 -gencode arch=compute_62,code=sm_62 for nvcc flags
Any suggestions?

BingoGit23, 2024-08-09:
解决方法：
注释掉src/darknet_ros/darknet_ros/CMakeLists.txt的189行
target_link_libraries(${PROJECT_NAME}
  ${PROJECT_NAME}_lib
# ${PROJECT_NAME}_core_lib
)
因为CUDA_FOUND 为真，不会编译darknet_ros_core_lib库

BingoGit23, 2024-08-09:
解决方法
注释src/darknet_ros/darknet_ros/CMakeLists.txt文件189行
target_link_libraries(${PROJECT_NAME}
  ${PROJECT_NAME}_lib
#${PROJECT_NAME}_core_lib
)
因为 CUDA_FOUND 为真，不会编译darknet_ros_core_lib库