leggedrobotics/darknet_ros #345: Cannot run yolov4-tiny while yolov3-tiny works ok for jetson nano

kyuhyong, 2021-08-03:
**Describe the bug**
I was trying to detect an object with custom trained yolo weights. 
I trained weight using https://github.com/AlexeyAB/darknet on PC with GPU without any problem.

For this, I used my own data set "my_data"
https://drive.google.com/file/d/1GcS9z2oqR1z5DuZpuAP-qEdJ7O8gntbx/view?usp=sharing

### For yolov3-tiny
First I tried with yolov3-tiny by getting weight
$ wget https://pjreddie.com/media/files/yolov3-tiny.weights
then made pre-trained weight by
$ ./darknet partial cfg/yolov3-tiny.cfg yolov3-tiny.weights yolov3-tiny.conv.15 15

Then train with following command
$ ./darknet detector train my_data/detector.data my_data/cfg/yolov3-tiny_mydata.cfg yolov3-tiny.conv.15 -map >> log/my_tiny_data.log

### For yolov4-tiny
Download weight from 
https://github.com/AlexeyAB/darknet/releases/download/darknet_yolo_v4_pre/yolov4-tiny.weights

Then made pre-trained wieght by
$ ./darknet partial cfg/yolov4-tiny.cfg yolov4-tiny.weights yolov4-tiny.conv.15 15

Train with
$ ./darknet detector train my_data/detector.data my_data/cfg/yolov4-tiny_mydata.cfg yolov4-tiny.conv.15 -map >> log/my_tiny_data.log

I put both wegith, cfg file under yolo_network_config folder
I made yaml file for both yolov3-tiny and yolov4-tiny
### yaml file for yolov3-tiny
yolo_model:

  config_file:
    name: yolov3-tiny_cup.cfg
  weight_file:
    name: yolov3-tiny_cup_final.weights
  threshold:
    value: 0.7
  detection_classes:
    names:
      - cup
### yaml file for yolov4-tiny
yolo_model:

  config_file:
    name: yolov4-tiny_cup.cfg
  weight_file:
    name: yolov4-tiny_cup_final.weights
  threshold:
    value: 0.7
  detection_classes:
    names:
      - cup

**Screenshots**
While yolov3-tiny works just fine as screenshot below
[image]

my yolov4-tiny doesn't work with error like below. 
```
Unused field: 'scale_x_y = 1.05'
Unused field: 'cls_normalizer = 1.0'
Unused field: 'iou_normalizer = 0.07'
Unused field: 'iou_loss = ciou'
Unused field: 'resize = 1.5'
Unused field: 'nms_kind = greedynms'
Unused field: 'beta_nms = 0.6'
   31 route  27
   32 conv    128  1 x 1 / 1    13 x  13 x 256   ->    13 x  13 x 128  0.011 BFLOPs
   33 upsample            2x    13 x  13 x 128   ->    26 x  26 x 128
   34 route  33 23
   35 conv    256  3 x 3 / 1    26 x  26 x 384   ->    26 x  26 x 256  1.196 BFLOPs
   36 conv     18  1 x 1 / 1    26 x  26 x 256   ->    26 x  26 x  18  0.006 BFLOPs
   37 yolo
Unused field: 'scale_x_y = 1.05'
Unused field: 'cls_normalizer = 1.0'
Unused field: 'iou_normalizer = 0.07'
Unused field: 'iou_loss = ciou'
Unused field: 'resize = 1.5'
Unused field: 'nms_kind = greedynms'
Unused field: 'beta_nms = 0.6'
Loading weights from /home/r1mini/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov4-tiny_cup_final.weights...Done!
Waiting for image.
[ INFO:0] Initialize OpenCL runtime...
[darknet_ros-1] process has died [pid 12913, exit code -11, cmd /home/r1mini/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/main_camera/image_raw __name:=darknet_ros __log:=/home/r1mini/.ros/log/4191f940-f3f8-11eb-9662-38002506a2ca/darknet_ros-1.log].
log file: /home/r1mini/.ros/log/4191f940-f3f8-11eb-9662-38002506a2ca/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

What could have been wrong wit making weights? or trainig?

**System (please complete the following information):**
 - OS: ubuntu 18.04 Jetpack 4.4

$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020

 - ROS version: melodic
 - GPU when relevant: Jetson nano

zeta0707, 2021-08-08:
Hi,

I also see same problem on Jetpack4.5.1.

OS: ubuntu 18.04 Jetpack 4.5.1

Thanks!
[image]

DanielYG, 2021-08-11:
Hello, 

Same problem here, any possible solution?

Thanks

Ar-Ray-code, 2021-08-12:
This problem is similar to #328 .

zeta0707, 2021-08-14:
@Ar-Ray-code 
Yes, I'm now using https://github.com/Tossy0423/yolov4-for-darknet_ros.
It supports Yolo4, but opencv3.4.x is required.

Thanks!

lawliet1997, 2022-08-25:
I am having hard time publishing the image topic after i run the launch file.
What are you guys using to open up the camera? And how are you guys connecting your camera to Yolo?
I am using 
Nvidia Jetson Xavier
Raspberry Pie V2 Camera
Ubuntu 18 with Ros Melodic