leggedrobotics/darknet_ros #350: Conflicting declaration of C function `void show_image_cv` when building for ROS2 Foxy

mnissov, 2021-09-17:
**Describe the bug**
Building the `foxy` branch of `darknet_ros` results in several errors
* `/home/USER/ros2_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:62:17: error: conflicting declaration of C function ‘void show_image_cv(image, const char*, IplImage*)’`
* /home/USER/ros2_ws/src/darknet_ros/darknet/examples/attention.c:290:9: error: too few arguments to function ‘show_image’

Have attempted building for Noetic instead, where it works with no problems.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd ~/ros_ws/src`
2. `git clone --recursive git@github.com:leggedrobotics/darknet_ros.git`
3. `git fetch && git checkout foxy`
4. `cd ../`
5. `colcon build --symlink-install --packages-select darknet_ros darknet_ros_msgs --cmake-args -DCMAKE_BUILD_TYPE=Release`

**System (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS version: Foxy
 - GPU when relevant: Nvidia GTX 1650 Ti

**Additional context**
Add any other context about the problem here.

Ar-Ray-code, 2021-09-25:
Please tell me OpenCV-version.

For OpenCV4, the submodule darknet is not available, so please consider the following repository.

[darknet-fp16 (using AlexeyAB/darknet)](https://github.com/Ar-Ray-code/darknet_ros_fp16)

takeyamayuki, 2023-03-24:
Same problem, same environments here. 

## error

```bash
$ colcon build --symlink-install --packages-select darknet_ros --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> darknet_ros
--- stderr: darknet_ros                            
/home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c: In function ‘validate_attention_single’:
/home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c:290:9: error: too few arguments to function ‘show_image’
  290 |         show_image(crop, "crop");
      |         ^~~~~~~~~~
In file included from /home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c:1:
/home/frontieris/ros2_ws/src/darknet_ros/darknet/include/darknet.h:709:5: note: declared here
  709 | int show_image(image p, const char *name, int ms);
      |     ^~~~~~~~~~
/home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c:301:13: error: too few arguments to function ‘show_image’
  301 |             show_image(tile, "tile");
      |             ^~~~~~~~~~
In file included from /home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c:1:
/home/frontieris/ros2_ws/src/darknet_ros/darknet/include/darknet.h:709:5: note: declared here
  709 | int show_image(image p, const char *name, int ms);
      |     ^~~~~~~~~~
make[2]: *** [CMakeFiles/darknet_ros_lib.dir/build.make:2711: CMakeFiles/darknet_ros_lib.dir/home/frontieris/ros2_ws/src/darknet_ros/darknet/examples/attention.c.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
In file included from /home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:27,
                 from /home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/opt/ros/foxy/include/image_transport/image_transport.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                                                         ^
In file included from /home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:62:17: error: conflicting declaration of C function ‘void show_image_cv(image, const char*, IplImage*)’
   62 | extern "C" void show_image_cv(image p, const char *name, IplImage *disp);
      |                 ^~~~~~~~~~~~~
In file included from /home/frontieris/ros2_ws/src/darknet_ros/darknet/src/network.h:6,
                 from /home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:49,
                 from /home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/home/frontieris/ros2_ws/src/darknet_ros/darknet/src/image.h:20:5: note: previous declaration ‘int show_image_cv(image, const char*, int)’
   20 | int show_image_cv(image im, const char* name, int ms);
      |     ^~~~~~~~~~~~~
/home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp: In function ‘image** darknet_ros::load_alphabet_with_file_cp(char*)’:
/home/frontieris/ros2_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:578:18: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
  578 |   char* labels = "/labels/%d_%d.png";
      |                  ^~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/darknet_ros_lib.dir/build.make:2087: CMakeFiles/darknet_ros_lib.dir/src/YoloObjectDetector.cpp.o] エラー 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/darknet_ros_lib.dir/all] エラー 2
make: *** [Makefile:141: all] エラー 2
---
Failed   <<< darknet_ros [2.66s, exited with code 2]

Summary: 0 packages finished [2.81s]
  1 package failed: darknet_ros
  1 package had stderr output: darknet_ros
```

## environments

- Ubuntu 20.04
- ROS foxy
- NVIDIA RTX 3080
- CUDA 11.8
```bash
$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0
```
- OpenCV 4.2.0
```bash
$ apt show libopencv-dev
...
Version: 4.2.0+dfsg-5
...
```

takeyamayuki, 2023-03-24:
Perhaps it is due to different reference commits in `darknet`.
I'm not sure which commit to reference.

poeldan, 2023-06-14:
Is there already a solution? I am facing the same problem

System:
    OS: Ubuntu 20.04
    ROS version: Humble
    GPU: Nvidia GeForce GTX 1060
    OpenVC: 4.5.4

Savarati, 2023-12-13:
darknet_ros/darknet under master and foxy branch are  different git repopositories

We can do that:
cd darkent_ros
rm -rf darkent
git clone https://github.com/pjreddie/darknet.git
git reset --hard 508381b37fe75e0e1a01bcb2941cb0b31eb0e4c9

Also note, opencv version used is 4.2.0

I use:
Ubuntu 22.04  
ROS Humble
compile successfully