leggedrobotics/darknet_ros #352: ROS catkin_make error

lingogogo, 2021-09-27:
error output in terminal

```
/home/gordonlin/robot/devel/lib/libdarknet_ros_lib.so: undefined reference to `pull_implicit_layer'
/home/gordonlin/robot/devel/lib/libdarknet_ros_lib.so: undefined reference to `ipl_into_image'
/home/gordonlin/robot/devel/lib/libdarknet_ros_lib.so: undefined reference to `push_implicit_layer'
/home/gordonlin/robot/devel/lib/libdarknet_ros_lib.so: undefined reference to `ipl_to_image'
/home/gordonlin/robot/devel/lib/libdarknet_ros_lib.so: undefined reference to `make_implicit_layer'
collect2: error: ld returned 1 exit status
darknet_ros_yolov4-master/darknet_ros/CMakeFiles/darknet_ros.dir/build.make:187: recipe for target '/home/gordonlin/robot/devel/lib/darknet_ros/darknet_ros' failed
make[2]: *** [/home/gordonlin/robot/devel/lib/darknet_ros/darknet_ros] Error 1
CMakeFiles/Makefile2:8202: recipe for target 'darknet_ros_yolov4-master/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed
make[1]: *** [darknet_ros_yolov4-master/darknet_ros/CMakeFiles/darknet_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

CMakelist.txt in the darknet_ros

# Find rquired packeges
find_package(Boost REQUIRED COMPONENTS thread)
find_package(OpenCV 4.1.1 REQUIRED)

find_package(OpenCV REQUIRED)
```
include_directories(${OpenCV_INCLUDE_DIRS})
message("wocako ${OpenCV_INCLUDE_DIRS}")
find_package(catkin REQUIRED
  COMPONENTS
    cv_bridge
    roscpp
    rospy
    std_msgs
    actionlib
    darknet_ros_msgs
    image_transport
)
```

I used opencv4.1.1.
The first problem is error 4.x cannot support c++11
but i solve the problem by downloading opencv3.4.1.
However , there are more undefined reference ,so i take the 4.1.1 version again.
there is no error with c++11
The error was the same with the opencv3.4.1.

`undefined referece to xxxxx`

but i think darknet using opencv 4.1.1 suitably.

Thank a lot in advance.

hamzaMahdi, 2022-03-08:
I have the same issue. Did you figure out a solution @lingogogo ? 
Much appreciated.

wlgnsthd, 2022-07-12:
I have the same issue too

CraftsionBoo, 2023-02-14:
I solved the problem at this link. about Q19 20 [https://www.cxyzjd.com/article/weixin_45702256/112464457](url)

hamdykhfagy, 2023-10-29:
I have the same issue