leggedrobotics/darknet_ros #361: OpenCV 4.x+ requires enabled C++11 support ROS2 Foxy

blakermchale, 2021-11-24:
**Describe the bug**
When I build the foxy branch of `darknet_ros` with CUDA support I get the following errors:
```
/usr/include/opencv4/opencv2/core/cvdef.h:704:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
  704 | #  error "OpenCV 4.x+ requires enabled C++11 support"
      |    ^~~~~
/usr/include/opencv4/opencv2/core/cvdef.h:710:10: fatal error: array: No such file or directory
  710 | #include <array>
```

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the foxy branch
2. `colcon build -DCMAKE_BUILD_TYPE=Release`
3. See error

**Expected behavior**
I expect ROS2 Foxy to compile with CUDA support.

**System (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS version: Foxy
 - GPU when relevant: NVIDIA RTX 2080 SUPER

**Additional context**
I noticed that this problem was fixed for Noetic in #308.

blakermchale, 2021-11-24:
Just saw #257 and the discussion at the end that referenced this fork https://github.com/ajaypaul2008/darknet_ros/tree/foxy

Ar-Ray-code, 2021-12-04:
Please try to use [my fork](https://github.com/Ar-Ray-code/darknet_ros_fp16) :) .

blakermchale, 2021-12-04:
Thanks! I switched to your branch on the PR you made and it works well!

Ar-Ray-code, 2021-12-08:
Please close this.

blakermchale, 2021-12-08:
Closed since #319 will fix this once it is merged.

closed 2021-12-08 by blakermchale