leggedrobotics/darknet_ros #363: Unsupported gpu architecture 'compute_30'

grafoteka, 2021-12-01:
**Describe the bug**
Error compiling darknet with CUDA drivers.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and install last version of CUDA drivers: 
```
$ wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/cuda-ubuntu2004.pinsudo 
$ mv cuda-ubuntu2004.pin /etc/apt/preferences.d/cuda-repository-pin-600
$ wget https://developer.download.nvidia.com/compute/cuda/11.5.1/local_installers/cuda-repo-ubuntu2004-11-5-local_11.5.1-495.29.05-1_amd64.deb
$ sudo dpkg -i cuda-repo-ubuntu2004-11-5-local_11.5.1-495.29.05-1_amd64.deb
$ sudo apt-key add /var/cuda-repo-ubuntu2004-11-5-local/7fa2af80.pub
$ sudo apt-get update
$ sudo apt-get -y install cuda
```
2. Compile ROS package
`$ catkin_make -DCMAKE_BUILD_TYPE=Release`
3. See error:
```
CMake Error at darknet_ros_lib_generated_col2im_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_col2im_kernels.cu.o

nvcc fatal   : Unsupported gpu architecture 'compute_30'

nvcc fatal   : Unsupported gpu architecture 'compute_30'
nvcc fatal   : Unsupported gpu architecture 'compute_30'
CMake Error at darknet_ros_lib_generated_blas_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_blas_kernels.cu.o

nvcc fatal   : Unsupported gpu architecture 'compute_30'
nvcc fatal   : Unsupported gpu architecture 'compute_30'
CMake Error at darknet_ros_lib_generated_dropout_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_dropout_layer_kernels.cu.o

CMake Error at darknet_ros_lib_generated_avgpool_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_avgpool_layer_kernels.cu.o

CMake Error at darknet_ros_lib_generated_deconvolutional_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_deconvolutional_kernels.cu.o

CMake Error at darknet_ros_lib_generated_convolutional_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_convolutional_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:65: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_activation_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:107: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_col2im_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:114: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_im2col_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:93: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_blas_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:100: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_dropout_layer_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:79: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_avgpool_layer_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:121: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_convolutional_kernels.cu.o] Error 1
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:86: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_deconvolutional_kernels.cu.o] Error 1
nvcc fatal   : Unsupported gpu architecture 'compute_30'
CMake Error at darknet_ros_lib_generated_crop_layer_kernels.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jorge/ws/ros1/darknet_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_crop_layer_kernels.cu.o

make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:72: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_crop_layer_kernels.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2597: darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed
```

**Expected behavior**
Compile corretly

**System (please complete the following information):**
 - Laptop DELL Precision 3561
 - OS: Ubuntu 20.04
 - ROS version: Noetic
 - GPU when relevant: NVidia T1200 Mobile

Ar-Ray-code, 2021-12-04:
Please try to delete `-gencode arch=compute_30,code=sm_30` on [CMakeLists.txt](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/CMakeLists.txt#L26).

It's already been removed in [my fork](https://github.com/Ar-Ray-code/darknet_ros_fp16).

aled96, 2021-12-10:
Try to remove all the `arch=compute_30,code=sm_30` in the CMakeLists.txt ad insert only:

`arch=compute_75,code=sm_75`

saqib772, 2023-11-26:
> Please try to delete `-gencode arch=compute_30,code=sm_30` on [CMakeLists.txt](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/CMakeLists.txt#L26).
> 
> It's already been removed in [my fork (ROS-Foxy)](https://github.com/Ar-Ray-code/darknet_ros_fp16).

there is no such thing in Cmakelists.txt ?