leggedrobotics/darknet_ros #366: Error in ' roslaunch darknet_ros darknet_ros.launch, ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)

mf093087, 2021-12-07:
After I set up the environment, I run this command 

**roslaunch darknet_ros darknet_ros.launch**

However, I met errors. 
[image]

Below is the full info of my code

mf093087@ubuntu:~/catkin_workspace$ roslaunch darknet_ros darknet_ros.launch
... logging to /home/mf093087/.ros/log/8562a1ca-57a2-11ec-8717-000c29fad970/roslaunch-ubuntu-10197.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:35581/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /home/mf093087/ca...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /camera/rgb/image...
 * /darknet_ros/weights_path: /home/mf093087/ca...
 * /darknet_ros/yolo_model/config_file/name: yolov2-tiny.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov2-tiny.weights
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

auto-starting new master
process[master]: started with pid [10207]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8562a1ca-57a2-11ec-8717-000c29fad970
process[rosout-1]: started with pid [10218]
started core service [/rosout]
ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Ishank-srivastava, 2022-01-24:
Did you find the solution? @mf093087

Roswellii, 2022-02-08:
I tried to run YOLO on ROS melodic too and I met the same problem. Does it have something to do with ROS version?

mf093087, 2022-02-08:
Yes, just try to use chmod +x +python file to give permission

AZ-Ismail, 2022-02-11:
> Yes, just try to use chmod +x +python file to give permission

I can't find the python file in the package. Can you post the directory?

Roswellii, 2022-02-16:
> 

Hi! I just solved this problem. I forgot to source the /devel/setup.bash again after compile the workspace.  Maybe you can try to source it again?

asibarr2, 2022-03-31:
I found a solution (3/31/22). 

Setup: Ubuntu 18.04
ROS Version: Melodic

So after you setup your keys, you are instructed to use the following command to build your package: 

git clone --recursive git@github.com:leggedrobotics/darknet_ros.git

^^ This command will not work for ROS Melodic users. You will get as far as the "cannot launch node of type" error with this build. So how I fixed this was downloading the feature/ci-melodic-and-noetic branch. As soon as I built this version in my catkin_ws/src directory, I sourced my workspace and launched a ros command and boom, it worked. So type this command to build the package for Ubuntu 18.04:

git clone --recursive git@github.com:leggedrobotics/darknet_ros.git --branch feature/ci-melodic-and-noetic

I hope this helps! From a fellow stressed out graduate student tired of all these damn bugs on this damn platform lol

ni-r1, 2022-04-01:
@asibarr2 This did not solve the problem!!, the error is persistent!!

lhakim85, 2022-04-15:
> @asibarr2 This did not solve the problem!!, the error is persistent..... I tried all the methods available on the internet but unable to solve this error, @asibarr2 if you find any other alternative ... Please do let me know

In my case, after catkin_make, i do rosdep install --from-paths src/ --ignore-src, and success

hello50505, 2022-05-26:
i face the same issue，have you fix it?

Roswellii, 2022-05-27:
> i face the same issue，have you fix it?

> Hi! My problem was simply due to I forgot to source setup.py in /dev folder of this project... I‘m not sure if we have the some problem. Could you provide more details?

Roswellii, 2022-05-27:
> i face the same issue，have you fix it? @[hello50505](https://github.com/hello50505)

> Hi! My problem was simply due to I forgot to source setup.py in /dev folder of this project... I‘m not sure if we have the some problem. Could you provide more details?

ni-r1, 2022-05-30:
I found a solution 

I created another directory and then installed the necessary libraries, later when I built the file...it worked out!!

kiranabbili, 2023-01-28:
ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure the file exists in the package path and permission is set to executable (chmod +x).

kiranabbili, 2023-01-28:
I tried many ways but I am unable to find the solution. can anyone help me.

ni-r1, 2023-01-28:
This issue can only be solved by creating a new directory and then deleting the previous libraries installed. now, start all over again by installing the libraries into this new directory! this is how I solved the problem 

ps: make sure that you've installed the correct version of libraries compactable with your version of python. and then you will not have any trouble building it!

AbbiliKiran, 2023-01-29:
@ni-r1 ,thank your reply. I just cloned the darknet_ros package and then I used this catkin_make -DCMAKE_BUILD_TYPE=Release. I have no idea about libraries. Do I need to anything else after cloning it.

Yasirsalam, 2023-02-01:
> 
Hello, Can you help me how you fixed this issue?

kiranabbili, 2023-02-01:
I am unable to find the solution. I don't know what is wrong.

Yasirsalam, 2023-02-02:
> setup.py

Hello, I didn't understand what you said exactly. Please help me out.

kiranabbili, 2023-02-15:
Can any one help me how to fix the error. ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure the file exists in the package path and permission is set to executable (chmod +x). I am using ROS-noetic. Thanks in advance.

NimishKashyap, 2023-03-27:
Is this issue still active?

kiranabbili, 2023-03-27:
I got the issue solved. The problem is due to failure of catkin_make due to error in some other packages. After deleting those packages, it worked.

NimishKashyap, 2023-03-27:
Can you elaborate on that? I'm still facing error even after creating a separate workspace

NimishKashyap, 2023-03-27:
And my catkin build is working fine without any failure

ParasIntern-SwapnilMathur, 2023-06-14:
@NimishKashyap Sir, How to solve this error in ubuntu 20.04?