leggedrobotics/darknet_ros #375: catkin build error OR How can i run the test?

ChrisTiberius, 2022-02-12:
I installed as described. 
Now i would like to test the package with:

christian@christian-P64-HJ-HK1:~/catkin_ws$ catkin build darknet_ros --no-deps --verbose --catkin-make-args run_tests
The build space at '/home/christian/catkin_ws/build' was previously built by 'catkin_make'. Please remove the build space or pick a different build space.

Did something go wrong with the installation? Any other way to quick test the package?

Thanks

**System (please complete the following information):**
 - OS: Ubuntu 18.04
 - ROS version: Melodic
 - GPU when relevant: NVidia GTX 1050TI

idochi, 2022-03-15:
I have the same problem. You can try to clean the workspace and then try it again, but for me it doesn't work either.

ppswaroopa, 2022-04-17:
Based on the command you're trying to run. Make sure to successfully build datknet_ros package using the [](https://github.com/leggedrobotics/darknet_ros#building)<br>
If you have previously built the workspace using catkin_make, make sure to delete the existing 'build' and 'devel' folders. Then build your workspace using the instructions in the link given.
Hope it helps.