leggedrobotics/darknet_ros #381: Runtime Issue - process has died pid 14258, exit code -11

panc8170, 2022-04-11:
I have followed the article to clone the git in the source folder of the catkin_workspace. The catkin_make command works perfectly fine for me. The issue creeps up when running following command:

roslaunch darknet_ros darknet_ros.launch 

ideally, in my other computers the node starts and works fine. But in the computer in question, I get the following output with error message highlighted.

... logging to /home/autodriveii/.ros/log/b300032c-b9b1-11ec-bce6-ed685309a8b3/roslaunch-autodriveii-X299X-AORUS-MASTER-14221.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:41077/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /home/autodriveii...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /image_publisher_...
 * /darknet_ros/weights_path: /home/autodriveii...
 * /darknet_ros/yolo_model/config_file/name: yolov2-tiny.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov2-tiny.weights
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

ROS_MASTER_URI=http://localhost:11311

process[darknet_ros-1]: started with pid [14258]
[ INFO] [1649693884.787531592]: [YoloObjectDetector] Node started.
[ INFO] [1649693884.791937168]: [YoloObjectDetector] Xserver is running.
[ INFO] [1649693884.793009708]: [YoloObjectDetector] init().
YOLO
layer     filters    size              input                output
    0 conv     16  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  16  0.150 BFLOPs
    1 max          2 x 2 / 2   416 x 416 x  16   ->   208 x 208 x  16
    2 conv     32  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  32  0.399 BFLOPs
    3 max          2 x 2 / 2   208 x 208 x  32   ->   104 x 104 x  32
    4 conv     64  3 x 3 / 1   104 x 104 x  32   ->   104 x 104 x  64  0.399 BFLOPs
    5 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    6 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128  0.399 BFLOPs
    7 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    8 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256  0.399 BFLOPs
    9 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
   10 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512  0.399 BFLOPs
   11 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024  1.595 BFLOPs
   13 conv    512  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x 512  1.595 BFLOPs
   14 conv    425  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 425  0.074 BFLOPs
   15 detection
mask_scale: Using default '1.000000'
Loading weights from /home/autodriveii/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov2-tiny.weights...Done!
Waiting for image.
**_[darknet_ros-1] process has died [pid 14258, exit code -11, cmd /home/autodriveii/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/image_publisher_1649693822022886118/image_raw __name:=darknet_ros __log:=/home/autodriveii/.ros/log/b300032c-b9b1-11ec-bce6-ed685309a8b3/darknet_ros-1.log].
log file: /home/autodriveii/.ros/log/b300032c-b9b1-11ec-bce6-ed685309a8b3/darknet_ros-1*.log_**
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

**System:**
 - OS: Ubuntu 20.04
 - ROS version: Noetic
 - GPU when relevant: No GPU

What am I missing here? How can I diagnose the problem?

AlexandreDuval, 2022-05-18:
Hi, I have the exact same issue for both `darknet.launch` and `yolo_v3.launch`

System:
- OS: Ubuntu 20.04
- ROS version: Noetic
- GPU when relevant: No GPU

Thank you for the help.

MRSHIMMER, 2022-05-20:
Try to delete Boost.

AlexandreDuval, 2022-05-20:
> Try to delete Boost.

Hi, thank you for your reply.
But, on my side, it does not work...

MRSHIMMER, 2022-05-20:
> > Try to delete Boost.
> 
> Hi, thank you for your reply. But, on my side, it does not work...

That's kind of bad. Today I reinstalled my Ubuntu and ROS, but failed to test Darknet_ROS using the USB_CAM package and my laptop camera. I thrashed about all day, but nothing came of it. Then I succeeded in my classmate's computer. But he doesn't install the Boost dependency as instructed. I deleted Boost and it worked. Our ubuntu versions are all Ubuntu18.04. Were you successful in printing camera data topics?

AlexandreDuval, 2022-05-20:
> 

Yes, my topics are good. I will try an other way. Thanks again.

MRSHIMMER, 2022-05-20:
> Yes, my topics are good. I will try an other way. Thanks again.

> 

> Yes, my topics are good. I will try an other way. Thanks again.

Wish you succes.

hello50505, 2022-05-26:
have you success?

MRSHIMMER, 2022-05-26:
> have you success?

Yes, you can try. Or you can learn about this library: [https://github.com/osrf/tensorflow_object_detector](url)

hello50505, 2022-05-26:
这是来自QQ邮箱的假期自动回复邮件。zzu水下空中机器人实验室电控组已经收到了你的提交。

AlexandreDuval, 2022-05-26:
I have **resolved** my issue.
I changed the `enable_opencv: true` to `enable_opencv: false` in the ros.yaml configuration file.

Hope it help.

cheexuu, 2023-01-06:
Try to use darkner_ros_gdb.launch to find what error is. If the information is shown like in   [#247](https://github.com/leggedrobotics/darknet_ros/issues/247). Then, this problem may be caused by the conflict or incompatibility of boost version. You can try to delete boost, catkin_make your project again, and run launch file.

BTW: check the file path of boost and remove them. The version of my installed boost is 1.81
deleting boost command:
`sudo rm -f /usr/local/lib/libboost*
sudo rm -rf /usr/local/include/boost
sudo rm -rf /usr/local/lib/cmake/*-1.81.0*`

Vaibhav2181, 2023-09-14:
Hello,
 It worked for me when i changed

from enable_opencv: true
to enable_opencv: false

in  darknet_ros/darknet_ros/config/ros.yaml file.