leggedrobotics/darknet_ros #382: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

joe28965, 2022-04-20:
**Describe the bug**
After building darknet_ros following the steps in the README, when I try to launch darknet_ros (roslaunch darknet_ros darknet_ros.launch) I get the following error the second it receives the image feed:

(darknet_ros:22136): Gtk-ERROR **: 11:44:10.920: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the build steps in the README
2. roslaunch darknet_ros darknet_ros.launch
3. Error

**Expected behavior**
I would expect darknet to start using the image feed.

**System (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS version: Noetic
 - GPU when relevant: NVidia AGX Xavier

joe28965, 2022-04-21:
Problem is in the image viewer. Disabling that 'fixes' the problem (at least for me).

lu-zilong, 2022-07-08:
I have the same problem.
Could you please tell how to solve it?
THANKS!

YHuniv, 2023-04-26:
Facing same problem, anyone have the solution ?
Thanks