leggedrobotics/darknet_ros #404: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

Pham-Canh-An-Tien, 2023-01-06:
**Describe the bug**
I am facing the bug using opencv-4.1.0 with darknet_ros. I understand that they are using different version on gtk+. Is there a workaround on this issue?

(YOLO:176179): Gtk-ERROR **: 17:12:20.981: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
[darknet_ros-1] process has died [pid 176179, exit code -5, cmd /home/learnlab/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/learnlab/.ros/log/a24096b6-8e02-11ed-9df0-c309f58de8de/darknet_ros-1.log].
log file: /home/learnlab/.ros/log/a24096b6-8e02-11ed-9df0-c309f58de8de/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit

Bruce-Si, 2023-03-06:
Hello, any progress? I met the same error. Though I compiled opencv and darknet_ros with `-D WITH_GTK_2_X=ON`, it doesn't work.

YHuniv, 2023-05-04:
Check your opencv versions, in my case i had opencv4.4 and 4.2 installed in same time, uninstalled 4.4 and it works.
Hope it helps !

Emre-SabT, 2023-10-04:
> Check your opencv versions, in my case i had opencv4.4 and 4.2 installed in same time, uninstalled 4.4 and it works. Hope it helps !

Hi, I have same error. How you check your opencv versions.  Can you explain your solution with steps please ? :)

Emre-SabT, 2023-10-06:
In my case, i changed the enable_opencv= true to false in darknet_ros/config/ros.yaml file and the error is gone, hope it helps.

YHuniv, 2023-10-10:
for Opencv version use interminal :
python3

>>> import cv2
>>> print(cv2.__version__)
4.2.0

If you put openCv=false i think i will be too slow , try this installation in this link if still need help.
https://www.youtube.com/watch?v=whAFl-izD-4&ab_channel=s1n7ax