leggedrobotics/darknet_ros #431: Extremely low fps

Lucas191105-Nanda, 2025-02-07:
**Describe the bug**
I already can use yolo with darknet ros. But when i run it on robot, with the specification of intel nuc 7 i5, i got extremely low fps, 1 fps. I already did everything right and did everything i can about the configuration, but none of it work. Can you suggest me the solution?

**Expected behavior**
I expect to get at least 20 fps when i run the yolo

**System (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS version: ROS Noetic
 - GPU when relevant: Intel HD Graphics 620

1508189250, 2025-02-28:
I also measured 1ftp around, I took a look at the time mainly in this function：float* prediction = network_predict(net_, X);I feel like I need cuda to speed it up

Lucas191105-Nanda, 2025-04-27:
> I also measured 1ftp around, I took a look at the time mainly in this function：float* prediction = network_predict(net_, X);I feel like I need cuda to speed it up

what hardware that you use specifically?